carla-simulator/ros-bridge #277: ROS2: port carla_ros_bridge

fpasch, 2020-04-30:


togaen, 2020-07-13:
Any status updates on this effort?

ruffsl, 2020-09-18:
While porting the bridge to ROS2, would it be feasible to use something other than python for bridging between carla's server and ROS2/DDS? For conducting simulations of robotics systems with high bandwidth, low latency requirements, using the rclcpp client library would seem more appropriate for performance gains and potential for shared memory/zero-copy IPC transport.

https://carla.readthedocs.io/en/latest/core_world/#client-creation 

Does Carla have a C++ API for it's server side TCP client access? Is it possible to use something like pluginlib to write a ROS2 bridge plugin that could run in the same process hosting the Carla server, skipping the intermediary TCP serialization altogether?

closed 2021-03-04 by fpasch