carla-simulator/ros-bridge #288: ROS2 Support

fpasch, 2020-04-30:
### Porting Priorities
(from high to low; topic describes milestone, that can be reached when porting the nodes below)

#### Setup for manual driving
1. [carla_ros_bridge](https://github.com/carla-simulator/ros-bridge/issues/277)
2. [carla_ego_vehicle](https://github.com/carla-simulator/ros-bridge/issues/276)
3. [carla_manual_control](https://github.com/carla-simulator/ros-bridge/issues/279)

#### Automatically follow precalculated route
4. [carla_ad_agent](https://github.com/carla-simulator/ros-bridge/issues/274)

#### Automatically follow generated waypoints
5. [carla_waypoint_publisher](https://github.com/carla-simulator/ros-bridge/issues/284)

#### Enable carla_ad_demo with rviz-plugin (execute scenarios, manual control, move camera)
6. [carla_ros_scenario_runner](https://github.com/carla-simulator/ros-bridge/issues/280)
7. [carla_spectator_camera](https://github.com/carla-simulator/ros-bridge/issues/281)
8. [carla_twist_to_control](https://github.com/carla-simulator/ros-bridge/issues/282)
9. [rviz_carla_plugin](https://github.com/carla-simulator/ros-bridge/issues/287)
10. [carla_ad_demo](https://github.com/carla-simulator/ros-bridge/issues/275)

#### Additional functionality
11. [carla_infrastructure](https://github.com/carla-simulator/ros-bridge/issues/278)
12. [carla_ackermann_control](https://github.com/carla-simulator/ros-bridge/issues/273)
13. [carla_walker_agent](https://github.com/carla-simulator/ros-bridge/issues/283)
14. [rqt_carla_plugin](https://github.com/carla-simulator/ros-bridge/issues/285)
15. [pcl_recorder](https://github.com/carla-simulator/ros-bridge/issues/286)

ll7, 2020-08-14:
Will the ros1 and ros2 bridge be within the same repository?

fpasch, 2020-08-14:
@ll7 Yes, that's the current plan. The ros2 branch is planned to become the new master (as it supports ros1 and ros2 in parallel)

anand-bala, 2021-02-09:
Hey @fpasch, I know this issue is old but I was wondering if the `ros2` branch is usable (unstable is fine)?

fpasch, 2021-02-10:
@anand-bala Yes. It should be pretty stable. (Testers are always welcome :-) )

anand-bala, 2021-02-10:
Thanks! Looks like you have a tester now 😄

closed 2021-03-26 by joel-mb