carla-simulator/ros-bridge #296: timeout issue for carla 0.9.9

spiegelss, 2020-05-05:
This is referencing [issue 272](https://github.com/carla-simulator/ros-bridge/issues/272) where a timeout occurred when waiting for world info.  This was closed yesterday and the master branch was updated.  However, I'm still getting the following timeout error 
```
[ERROR] [1588687463.809989, 0.000000]: Error: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has finished cleanly
log file: /home/sspiegel/.ros/log/503783be-8ed9-11ea-9d08-74e5f9befa2d/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
```
When I looked at the carla simulator It gives the following message when running it in the background

```
[2020.05.05-14.27.07:324][  0]LogRHI: Warning: Failed to find entry point for vkDestroySurfaceKHR
[2020.05.05-14.27.07:324][  0]LogRHI: Warning: Failed to find entry point for vkGetPhysicalDeviceSurfaceSupportKHR
[2020.05.05-14.27.07:324][  0]LogRHI: Warning: Failed to find entry point for vkGetPhysicalDeviceSurfaceCapabilitiesKHR
[2020.05.05-14.27.07:324][  0]LogRHI: Warning: Failed to find entry point for vkGetPhysicalDeviceSurfaceFormatsKHR
[2020.05.05-14.27.07:324][  0]LogRHI: Warning: Failed to find entry point for vkGetPhysicalDeviceSurfacePresentModesKHR
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=414640 LargeMemoryPoolOffset=545776 
```
which seems to indicate a memory issue.  I don't run into this when I run carla normally.  I issued the following command to run carla in the background:
```SDL_VIDEODRIVER=offscreen ./start.sh -opengl```
where **start.sh** is something I wrote in my carla folder and is 
```bash
#!/usr/bin/env bash
cd Unreal/CarlaUE4
~/UnrealEngine_4.24/Engine/Binaries/Linux/UE4Editor "$PWD/CarlaUE4.uproject"
```
(It's similar to the **CarlaU4.sh**).  
Am I correct in assuming it's a memory issue?  My computer has 32GB of RAM and 6GB of GPU memory.  Thanks!

fpasch, 2020-05-06:
This might be related to a missing vulkan driver.
Please check if running CarlaUE4.sh without parameters reports an error dialog.

If yes, please install the vulkan driver, e.g. by executing

    sudo apt install nvidia-driver-435

spiegelss, 2020-05-06:
I was able to start carla for a few seconds with the CarlaUE4.sh command, but it did crash with the memory error.  I've never had that problem when doing the ```make launch``` command before, which is curious.  

As far as the vulkan drivers are concerned, I have Nvidia drver version 435 already installed.  See images attached.  

[image]
[image]

fpasch, 2020-06-15:
As this seems not to be ROS-related, please raise your question [here](https://github.com/carla-simulator/carla/issues).

closed 2020-06-15 by fpasch