carla-simulator/ros-bridge #297: Roslaunch in tutorial not working in version 0.9.9

leodario7, 2020-05-08:
I'm currently on version 0.9.9 of Carla, using the precompiled method.

I'm following the documentation and cannot get ros-bridge to work properly. I get the following output:

```
roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch
... logging to /home/user/.ros/log/5cd0efd4-910e-11ea-8739-1c4d7022cf68/roslaunch-user-desktop-22568.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://user-desktop:39889/

SUMMARY
========

PARAMETERS
 * /carla/ego_vehicle/role_name: ['hero', 'ego_veh...
 * /carla/fixed_delta_seconds: 0.05
 * /carla/host: localhost
 * /carla/port: 2000
 * /carla/synchronous_mode: False
 * /carla/synchronous_mode_wait_for_vehicle_control_command: True
 * /carla/timeout: 2
 * /carla_ego_vehicle_ego_vehicle/role_name: ego_vehicle
 * /carla_ego_vehicle_ego_vehicle/sensor_definition_file: /home/user/ca...
 * /carla_ego_vehicle_ego_vehicle/spawn_ego_vehicle: True
 * /carla_ego_vehicle_ego_vehicle/spawn_point: 
 * /carla_ego_vehicle_ego_vehicle/vehicle_filter: vehicle.*
 * /carla_manual_control_ego_vehicle/role_name: ego_vehicle
 * /rosbag_fname: 
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /use_sim_time: True

NODES
  /
    carla_ego_vehicle_ego_vehicle (carla_ego_vehicle/carla_ego_vehicle.py)
    carla_manual_control_ego_vehicle (carla_manual_control/carla_manual_control.py)
    carla_ros_bridge (carla_ros_bridge/bridge.py)
    goal_relay_ego_vehicle (topic_tools/relay)
    initialpose_relay_ego_vehicle (topic_tools/relay)

auto-starting new master
process[master]: started with pid [22578]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5cd0efd4-910e-11ea-8739-1c4d7022cf68
process[rosout-1]: started with pid [22591]
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [22594]
process[goal_relay_ego_vehicle-3]: started with pid [22599]
process[initialpose_relay_ego_vehicle-4]: started with pid [22600]
process[carla_ego_vehicle_ego_vehicle-5]: started with pid [22601]
process[carla_manual_control_ego_vehicle-6]: started with pid [22606]
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
[INFO] [1588930149.677057, 0.000000]: listening to server localhost:2000
[INFO] [1588930149.677936, 0.000000]: using vehicle filter: vehicle.*
[INFO] [1588930149.679041, 0.000000]: Waiting for CARLA world (topic: /carla/world_info)...
[INFO] [1588930149.791936, 0.000000]: Trying to connect to localhost:2000
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.9-1-g67684c5b 
WARNING: Simulator API version  = 0.9.9 
[INFO] [1588930149.821454, 0.000000]: synchronous_mode: False
[INFO] [1588930149.822281, 0.000000]: fixed_delta_seconds: 0.05
[INFO] [1588930149.836022, 0.000000]: Created Spectator(id=1, parent_id=None, type=spectator, prefix=spectator, attributes={})
[INFO] [1588930149.837566, 0.000000]: Created Traffic(id=2, parent_id=None, type=traffic.stop, prefix=traffic, attributes={})
[INFO] [1588930149.839463, 0.000000]: Created Traffic(id=3, parent_id=None, type=traffic.stop, prefix=traffic, attributes={})
[INFO] [1588930149.841199, 0.000000]: Created Traffic(id=4, parent_id=None, type=traffic.stop, prefix=traffic, attributes={})
[... 78 lines cut ...]
[INFO] [1588930149.930545, 0.000000]: Created TrafficLight(id=83, parent_id=None, type=traffic.traffic_light, prefix=traffic.traffic_light, attributes={})
[INFO] [1588930149.931546, 0.000000]: Created TrafficLight(id=84, parent_id=None, type=traffic.traffic_light, prefix=traffic.traffic_light, attributes={})
[INFO] [1588930150.076019, 666.395021]: CARLA world available. Spawn ego vehicle...
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.9-1-g67684c5b 
WARNING: Simulator API version  = 0.9.9 
[INFO] [1588930150.436058, 667.845021]: Created EgoVehicle(id=85, parent_id=None, type=vehicle.gazelle.omafiets, prefix=ego_vehicle, attributes={'driver_id': '0', 'sticky_control': 'true', 'color': '202,88,176', 'object_type': '', 'number_of_wheels': '2', 'role_name': 'ego_vehicle'})
[INFO] [1588930150.436745, 667.845021]: Created ObjectSensor(parent_id=85, prefix=ego_vehicle/objects)
[INFO] [1588930150.514618, 667.845021]: Created RgbCamera(id=86, parent_id=85, type=sensor.camera.rgb, prefix=ego_vehicle/camera/rgb/front, attributes={'slope': '0.88', 'chromatic_aberration_offset': '0.0', 'exposure_min_bright': '0.1', 'fstop': '1.4', 'sensor_tick': '0.05', 'calibration_constant': '16.0', 'image_size_x': '800', 'lens_circle_falloff': '5.0', 'exposure_speed_up': '3.0', 'exposure_compensation': '3.0', 'fov': '100', 'tint': '0.0', 'enable_postprocess_effects': 'true', 'lens_kcube': '0.0', 'exposure_speed_down': '1.0', 'lens_x_size': '0.08', 'white_clip': '0.04', 'motion_blur_min_object_screen_size': '0.1', 'toe': '0.55', 'blur_amount': '1.0', 'lens_k': '-1.0', 'lens_circle_multiplier': '0.0', 'motion_blur_intensity': '0.45', 'exposure_mode': 'manual', 'role_name': 'front', 'shutter_speed': '60.0', 'blade_count': '5', 'blur_radius': '0.0', 'shoulder': '0.26', 'min_fstop': '1.2', 'temp': '6500.0', 'lens_y_size': '0.08', 'black_clip': '0.0', 'chromatic_aberration_intensity': '0.0', 'image_size_y': '600', 'iso': '1200.0', 'focal_distance': '1000.0', 'exposure_max_bright': '2.0', 'gamma': '2.2', 'motion_blur_max_distortion': '0.35'})
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 22594, exit code -11, cmd /home/user/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/user/.ros/log/5cd0efd4-910e-11ea-8739-1c4d7022cf68/carla_ros_bridge-2.log].
log file: /home/user/.ros/log/5cd0efd4-910e-11ea-8739-1c4d7022cf68/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
[carla_manual_control_ego_vehicle-6] killing on exit
[carla_ego_vehicle_ego_vehicle-5] killing on exit
[initialpose_relay_ego_vehicle-4] killing on exit
[goal_relay_ego_vehicle-3] killing on exit
[carla_ros_bridge-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

leodario7, 2020-05-11:
I was not using the 0.9.9 version of PythoAPI, I was using a newer version 0.9.9-1-g67684c5b, and matching it with "quick start" version of Carla. After changing everything to 0.9.9 the problem was solved.

closed 2020-05-11 by leodario7

MichaelKlin, 2020-06-07:
@leodario7 , how did you changed version to 0.9.9?

leodario7, 2020-06-09:
> 
> 
> @leodario7 , how did you changed version to 0.9.9?

I just checked out to 0.9.9 version in the repository