carla-simulator/ros-bridge #298: Is Object sensor active in carla-ros-bridge - using carla 0.9.9

praveenputhettu, 2020-05-13:
Carla Version used:: 0.9.9
Is Object sensor active in carla-ros-bridge?
i am not getting any outputs from the object sensor.

joel-mb, 2020-05-14:
Hi @praveenputhettu !
I am currently using CARLA 0.9.9 and I can correctly retrieve the objects from the `/carla/objects` topic. How are you launching the ros bridge?

praveenputhettu, 2020-05-14:
Hello joel-mb

i used the following commands to run the server, non-player vehicles and ego-vehicle.

cd CARLA_0.9.9/
 ./CarlaUE4.sh 

cd CARLA_0.9.9/PythonAPI/examples
 ./spawn_npc.py -n 15

roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch

joel-mb, 2020-05-15:
Hi @praveenputhettu , I cannot reproduce the problem.

Is the topic `/carla/objects` being published empty or it is not published at all?

praveenputhettu, 2020-05-18:
Hello joel-mb,

Sorry for the late responses.
When i run rostopic echo /carla/objects i am getting a warning message ' no messages received and simulated time is active ' 
Also when i hz the topic i am getting 'no new message output'

So i checked the sensor config file in carla_ego_vehicle but i did not find any object sensor attached.
So i  added an object sensor like
            "type": "sensor.other.obstacle",
            "id": "objectSnsr1",
            "x": 0.0, "y": 0.0, "z": 0.0, "roll": 0.0, "pitch": 0.0, "yaw": 0.0,
            "distance": 6.0,
            "hit_radius": 1.0
i spawned other vehicles using ./spawn_npc.py -n 15 and went near another vehicle.

But still the same output is received as mentioned above.
Am i doing the correct way?

joel-mb, 2020-05-21:
Hi @praveenputhettu 
As far as I know, is not necessary to add that sensor in the configuration. If you take a look at the bridge source code (https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/bridge.py), lines 140-143, the pseudo sensor `ObjectSensor` is always used (this is the one publishing in the topic `carla/objects`). 

Make sure that you are using the last version of the ROS bridge (are you using the debian package?) and that the bridge_node is not crashing when launching it.

praveenputhettu, 2020-05-22:
Hello joel-mb,
OK.

i had installed ros brigde from github
and followed the steps that mentioned for For Developers,

Create a catkin workspace and install carla_ros_bridge package
#setup folder structure
mkdir -p ~/carla-ros-bridge/catkin_ws/src
cd ~/carla-ros-bridge
git clone https://github.com/carla-simulator/ros-bridge.git
cd ros-bridge
git submodule update --init
cd ../catkin_ws/src
ln -s ../../ros-bridge
source /opt/ros/kinetic/setup.bash
cd ..

#install required ros-dependencies
rosdep update
rosdep install --from-paths src --ignore-src -r

#build
catkin_make

Is there any issue?

joel-mb, 2020-05-25:
Hi @praveenputhettu. 
It should be working fine. I have tried again to reproduce the issue but it's working fine to me.
Can you run the bridge in this way and send me the resulting file?
```bash
roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch >> log.txt
```

Also check the topic `carla/ego_vechile/objects`. Apart from the global object sensor (published in `carla/objects`), each ego vehicle has also attached an object sensor (`carla/<role-name>/objects`) which is publishing all the objects in the Town.

praveenputhettu, 2020-05-27:
Hello joel-mb,

Following is the log created on running,
roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch >> log.txt

[log.txt](https://github.com/carla-simulator/ros-bridge/files/4688181/log.txt)

Both carla/ego_vechile/objects and carla/objects doesnt have any incoming data.

joel-mb, 2020-05-28:
Hmm, everything seems fine (I'm running the exact same configuration but using ROS melodic).

Please ensure that the clock message is being published: `rostopic echo /clock`. Do you see an image in the pygame window?

I'm not 100% sure, but I think this is not related to the `ObjectSensor`. Are you able to print any other message published by the ros bridge? E.g., `/carla/status`

praveenputhettu, 2020-05-29:
Hello joel-mb,

/clock is being published
everything else is working.

Do you got this warning message
Failed to serialize message on publishing: field objects[].classification_certainty must be unsigned integer type.

May be this is the reason. Do you know how to solve it.

joel-mb, 2020-06-08:
Hi @praveenputhettu 
Sorry for my late response.
As I have said, with the current version of master, I'm not able to reproduce the problem.

Can you try substituting line `102` of `traffic_participant.py` to:
```py
obj.classification_certainty = 255
```

Let me know if the problem persists.

praveenputhettu, 2020-06-09:
Thank you joel-mb 
its solved.
i had changed obj.classification_certainty = 1.0  to  obj.classification_certainty = 1

closed 2020-06-09 by praveenputhettu