carla-simulator/ros-bridge #300: Rosbridge process had died

lijecaru, 2020-05-27:
After following the installation steps (developer) I get the following error when running the launch file:

`(carla-rl) linda@linda:~$ roslaunch carla_ros_bridge carla_ros_bridge.launch
... logging to /home/linda/.ros/log/93f3f0d4-a071-11ea-9341-1831bf9f401a/roslaunch-linda-17214.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://linda:34035/

SUMMARY
========

PARAMETERS
 * /carla/ego_vehicle/role_name: ['hero', 'ego_veh...
 * /carla/fixed_delta_seconds: 0.05
 * /carla/host: localhost
 * /carla/port: 2000
 * /carla/synchronous_mode: False
 * /carla/synchronous_mode_wait_for_vehicle_control_command: True
 * /carla/timeout: 2
 * /rosbag_fname: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    carla_ros_bridge (carla_ros_bridge/bridge.py)

auto-starting new master
process[master]: started with pid [17224]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 93f3f0d4-a071-11ea-9341-1831bf9f401a
process[rosout-1]: started with pid [17237]
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [17248]
Traceback (most recent call last):
  File "/home/linda/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 24, in <module>
    **import carla**
ModuleNotFoundError: No module named 'carla'
================================================================================**REQUIRED process [carla_ros_bridge-2] has died!**
process has died [pid 17248, exit code 1, cmd /home/linda/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/linda/.ros/log/93f3f0d4-a071-11ea-9341-1831bf9f401a/carla_ros_bridge-2.log].
log file: /home/linda/.ros/log/93f3f0d4-a071-11ea-9341-1831bf9f401a/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
[carla_ros_bridge-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

Apparently, the system does not recognize the Carla module, I have exported the PYTHONPATH as required. I run everything within a conda environment, so I'm not sure if that interferes somehow with the launchfile.

The process is the following:

0. Run the Carla simulator in the background in other terminal
1. Activate conda environment
2. Export PYTHONPATH
3. Source the ROS bridge repo download: `source ~/carla-ros-bridge/catkin_ws/devel/setup.bash`
3. Start ros bridge: `roslaunch carla_ros_bridge carla_ros_bridge.launch`

joel-mb, 2020-05-28:
Hi @lijecaru !
I have never worked with conda so I don't know how it is working with the `PYTHONPATH` variable. My recommendation is that you add the `export PYTHONPATH=...` command into your ~/.bashrc. Make sure you indicate the path to the egg file correctly.

If this is not working you can always create a `carla.pth` file into your `{path-to-conda-env}/lib/python2.7/site-packages` folder adding the path to the egg file. The content of the `carla.pth` file should be something like this:

```
<path-to-carla>/PythonAPI/carla/dist/<egg-file>
<path-to-carla>/PythonAPI/carla/
```

lijecaru, 2020-05-28:
Thanks @joel-mb , I added the path to the egg file, but had to keep everything with python 2.7 for it to work. Any experience making it work with python 3.7?

joel-mb, 2020-05-29:
@lijecaru ROS kinetic and melodic use Python2 by default. This will change with the new release ROS Noetic which will target only Python3 and Ubuntu 20.04 (http://wiki.ros.org/noetic)

lijecaru, 2020-05-29:
Got it, thanks!

closed 2020-05-29 by lijecaru

gokay98, 2024-03-25:
> Thanks @joel-mb , I added the path to the egg file, but had to keep everything with python 2.7 for it to work. Any experience making it work with python 3.7?
Hi lijecaru. Can you write it down your egg file location.