carla-simulator/ros-bridge #302: Add compressed image publication to ROS bridge

ArtificialNebulae, 2020-06-01:
# Summary
When running CARLA over the network, camera imagery alone may require well over a gigabit, even with the default configuration. While the base CARLA simulator does not support compressed imagery transport, it is relatively easy to enable compressed image transport over the ROS bus. This would be intended for use cases where CARLA and the ROS bridge run on the same machine, with additional nodes such as the remote control run on a networked remote machine.

# Implementation Proposal
1. To `carla_ros_bridge/src/carla_ros_bridge/camera.py`, add a new publisher for compressed images if the camera has the `publish_compressed` set to `jpeg`, `jpg`, or `png`, with quality settings read from `compression_level` (0-100 for JPG, 0-9 for PNG). The image would be converted from the original image using `cv2.imencode(), then published to $CAMERA_PREFIX/compressed as a message of type `sensor_msgs/CompressedImage`.
1. To `carla_remote_control/src/carla_remote_control`, add a `view_compressed` parameter that changes the subscribed image to $CAMERA_PREFIX/compressed of type `sensor_msgs/CompressedImage.`
1. (Optional) Check if there are any subscribers to camera topic(s), and skip publication if there are no subscribers.

I have a rough prototype working in a private branch, if there is interest I can clean up the code and set up a PR.

fpasch, 2020-06-15:
Sounds good. Please feel free to create a PR.

hofbi, 2020-11-11:
@ArtificialNebulae If you are still looking for the compression feature you could try [TELECARA](https://github.com/hofbi/telecarla). This package includes a streaming node which uses the x264 video codec to stream the data over the network and publishes them on the client side. It supports live parameter adaptation using dynamic reconfigure to either set the quantization parameter or a target bitrate.

A second option if you really need the image compression would be to use the [image transport republish](http://wiki.ros.org/image_transport#Usage) node.