carla-simulator/ros-bridge #303: Lidar data not visible in rviz

vedthakur5, 2020-06-03:
I am using 
Carla: 0.9.9
Carla-ros -bridge: latest version and
Ubuntu:18.04
Ros: meloldic

this my launch file code:

<launch>
  
  <arg name='host' default='localhost'/>
  <arg name='port' default='2000'/>
  <arg name='timeout' default='2'/>

  
  <arg name='role_name' default='ego_vehicle'/>
  <arg name="vehicle_filter" default='vehicle.*'/>
  <arg name="spawn_point" default=""/>
  <arg name="sensor_definition_file" default="$(find carla_ego_vehicle)/config/sensors2.json"/>

  
  <arg name="town" default=''/>

  
  <arg name='synchronous_mode' default='True'/> 
  <arg name='synchronous_mode_wait_for_vehicle_control_command' default='False'/>
  <arg name='fixed_delta_seconds' default='0.05'/>

  <include file="$(find carla_ros_bridge)/launch/carla_ros_bridge.launch">
    <arg name='host' value='$(arg host)'/>
    <arg name='port' value='$(arg port)'/>
    <arg name='town' value='$(arg town)'/>
    <arg name='timeout' value='$(arg timeout)'/>
    <arg name='synchronous_mode' value='$(arg synchronous_mode)'/>
    <arg name='synchronous_mode_wait_for_vehicle_control_command' value='$(arg synchronous_mode_wait_for_vehicle_control_command)'/>
    <arg name='fixed_delta_seconds' value='$(arg fixed_delta_seconds)'/>
  </include>

  <include file="$(find carla_ego_vehicle)/launch/carla_example_ego_vehicle.launch">
    <arg name='host' value='$(arg host)'/>
    <arg name='port' value='$(arg port)'/>
    <arg name='timeout' value='$(arg timeout)'/>
    <arg name="vehicle_filter" value='$(arg vehicle_filter)'/>
    <arg name='role_name' value='$(arg role_name)'/>
    <arg name="spawn_point" value="$(arg spawn_point)"/>
    <arg name="sensor_definition_file" value='$(arg sensor_definition_file)'/>
  </include>

  <include file="$(find carla_manual_control)/launch/carla_manual_control.launch">
    <arg name='role_name' value='$(arg role_name)'/>
  </include>

  
  <node pkg="rviz" type="rviz" name="rviz" args="-d $(find carla_ros_bridge)/config/bro.rviz" required="true" output="screen"/>
  

</launch>

This is my sensors2.json file:

{
    "sensors": [
        {
            "type": "sensor.camera.rgb",
            "id": "front",
            "x": 2.0, "y": 0.0, "z": 2.0, "roll": 0.0, "pitch": 0.0, "yaw": 0.0,
            "width": 800,
            "height": 600,
            "fov": 100,
            "sensor_tick": 0.05,
[... 293 lines cut ...]
      Value: Orbit (rviz)
      Yaw: 0.895399808883667
    Saved: ~
Window Geometry:
  Displays:
    collapsed: false
  Height: 846
  Hide Left Dock: false
  Hide Right Dock: false
  Image:
    collapsed: false
  QMainWindow State: 000000ff00000000fd000000040000000000000156000002b0fc0200000009fb0000001200530065006c0065006300740069006f006e00000001e10000009b0000005c00fffffffb000000120056006900650077007300200054006f006f02000001df000002110000018500000122fb000000200054006f006f006c002000500072006f0070006500720074006900650073003203000002880000011d000002210000017afb0000001e0054006f006f006c002000500072006f0070006500720074006900650073010000003d000000a30000005c00fffffffb000000100044006900730070006c00610079007301000000e600000174000000c900fffffffb0000002000730065006c0065006300740069006f006e00200062007500660066006500720200000138000000aa0000023a00000294fb00000014005700690064006500530074006500720065006f02000000e6000000d2000003ee0000030bfb0000000c004b0069006e0065006300740200000186000001060000030c00000261fb0000000a0049006d00610067006501000002600000008d0000001600ffffff000000010000010f000002b0fc0200000003fb0000001e0054006f006f006c002000500072006f00700065007200740069006500730100000041000000780000000000000000fb0000000a00560069006500770073010000003d000002b0000000a400fffffffb0000001200530065006c0065006300740069006f006e010000025a000000b200000000000000000000000200000490000000a9fc0100000001fb0000000a00560069006500770073030000004e00000080000002e10000019700000003000004b00000003efc0100000002fb0000000800540069006d00650100000000000004b0000002eb00fffffffb0000000800540069006d006501000000000000045000000000000000000000023f000002b000000004000000040000000800000008fc0000000100000002000000010000000a0054006f006f006c00730100000000ffffffff0000000000000000
  Selection:
    collapsed: false
  Time:
    collapsed: false
  Tool Properties:
    collapsed: false
  Views:
    collapsed: false
  Width: 1200
  X: 67
  Y: 27

I am getting the camera sensors but i am not able to see lidar sensor output on rviz

vedthakur5, 2020-06-03:
I forgot to post the launch file
Launch file :

<launch>
  
  <arg name='host' default='localhost'/>
  <arg name='port' default='2000'/>
  <arg name='timeout' default='2'/>

  
  <arg name='role_name' default='ego_vehicle'/>
  <arg name="vehicle_filter" default='vehicle.*'/>
  <arg name="spawn_point" default=""/>
  <arg name="sensor_definition_file" default="$(find carla_ego_vehicle)/config/sensors2.json"/>

  
  <arg name="town" default=''/>

  
  <arg name='synchronous_mode' default='True'/> 
  <arg name='synchronous_mode_wait_for_vehicle_control_command' default='False'/>
  <arg name='fixed_delta_seconds' default='0.05'/>

  <include file="$(find carla_ros_bridge)/launch/carla_ros_bridge.launch">
    <arg name='host' value='$(arg host)'/>
    <arg name='port' value='$(arg port)'/>
    <arg name='town' value='$(arg town)'/>
    <arg name='timeout' value='$(arg timeout)'/>
    <arg name='synchronous_mode' value='$(arg synchronous_mode)'/>
    <arg name='synchronous_mode_wait_for_vehicle_control_command' value='$(arg synchronous_mode_wait_for_vehicle_control_command)'/>
    <arg name='fixed_delta_seconds' value='$(arg fixed_delta_seconds)'/>
  </include>

  <include file="$(find carla_ego_vehicle)/launch/carla_example_ego_vehicle.launch">
    <arg name='host' value='$(arg host)'/>
    <arg name='port' value='$(arg port)'/>
    <arg name='timeout' value='$(arg timeout)'/>
    <arg name="vehicle_filter" value='$(arg vehicle_filter)'/>
    <arg name='role_name' value='$(arg role_name)'/>
    <arg name="spawn_point" value="$(arg spawn_point)"/>
    <arg name="sensor_definition_file" value='$(arg sensor_definition_file)'/>
  </include>

  <include file="$(find carla_manual_control)/launch/carla_manual_control.launch">
    <arg name='role_name' value='$(arg role_name)'/>
  </include>

  
  <node pkg="rviz" type="rviz" name="rviz" args="-d $(find carla_ros_bridge)/config/bro.rviz" required="true" output="screen"/>
  

</launch>

vedthakur5, 2020-06-03:
I don't kow why the launch files is not being printed here so i have posted a pic:
[image]

fpasch, 2020-06-15:
Please compare with launch configuration carla_ad_demo_with_rviz.launch from carla_ad_demo

vedthakur5, 2020-06-17:
Ya thanks for your comment....actually my pointcloud topic was wrong...originally it is 
Topic: /carla/ego_vehicle/lidar/front/point_cloud
but it should be
Topic: /carla/ego_vehicle/lidar/lidar1/point_cloud
as the id of lidar sensor is lidar1 in sensors.json

closed 2020-06-17 by fpasch