carla-simulator/ros-bridge #305: Carla AD Demo Scenario: FollowLeadingVehicle Target is not moving

Roosstef, 2020-06-04:
Hey all,

when Im starting the FollowLeadingVehicle.osx Scenario in the carla_ad_demo.launch the target vehicle is not moving. When spawning normal npcs with spawn_npc.py they are moving. 
Is this a bug in the ros-bridge or a bug in the used FollowLeadingVehicle.xosc file?

This file also differs from the FollowLeadingVehicle.xosc I found in the scenario_runner repo.

Are there any guides to convert the xosc files to use them with  carla_ad_demo.launch?

fpasch, 2020-06-15:
It should move. But due to synchronous mode it might take some time.

closed 2020-06-15 by fpasch

dvh549, 2020-06-16:
Hi @fpasch, I tried executing the scenario and waited for the ego vehicle to move but it didn't. I tried multiple times but the scenario always fails at execution and status turns to red. However, the spawned npcs moved as per norm after awhile. Upon closer look, I found these errors in the terminal as shown in the images. Please advise on how to proceed.
[image]
[image]