carla-simulator/ros-bridge #307: Issue with IMU orientation?

rohanb2018, 2020-06-05:
I noticed that in the IMU [file](https://github.com/carla-simulator/ros-bridge/blob/d0b714c74128c7d135145559b43e123244ea9b5b/carla_ros_bridge/src/carla_ros_bridge/imu.py#L63-L68), the IMU orientation calculation doesn't seem to be producing values that are in the standard ROS (right-handed ENU) frame.

The solution that I came up with replaced the following line (which keeps the quaternion in the left-handed CARLA world frame):

```
quat = tf.transformations.quaternion_from_euler(math.radians(imu_rotation.roll),
                                                        math.radians(imu_rotation.pitch),
                                                        math.radians(imu_rotation.yaw))
```

with this line, which does the coordinate frame conversion properly (from the left-handed CARLA world frame, to the right-handed ROS ENU frame):

```
quat = trans.carla_rotation_to_numpy_quaternion(imu_rotation)
```

Can someone verify that this is actually an issue with the IMU? Is the IMU ROS message supposed to be in the left-handed CARLA world frame?

GabiShahmayster, 2020-06-18:
HI Rohan,

How are you?
Did you get a chance to get to the bottom of this issue?
I have another issue with Carla IMU data [(https://github.com/carla-simulator/carla/issues/2939)](url), which has not received any reply from the Carla team, yet. 
The IMU gyros data seems to be delivered in the world-frame, instead of the body-frame. On one hand, it seems very unlikely that such a huge bug would go unnoticed, on the other, they've recently fixed a bug which had gravity constantly added to the z-axis accelerometer, disregarding IMU orientation (version 0.9.9.1)...
Did you, by any chance, notice anything similar?

Thank you in advance,
Gabi Shahmayster

ItsTimmy, 2020-06-27:
If anyone else comes across this and is wondering about the gyro issue, there is a fix in this PR: https://github.com/carla-simulator/carla/pull/2998

heethesh, 2020-12-04:
Aren't the [angle conversions in transforms.py](https://github.com/carla-simulator/ros-bridge/blob/master/carla_common/src/carla_common/transforms.py#L109) also incorrect then? Pitch and yaw angles are negated, shouldn't the negation be for roll and yaw instead? Also see [this conversion here](https://github.com/code-iai/ROSIntegration/blob/master/Source/ROSIntegration/Private/TFBroadcastComponent.cpp#L144).