carla-simulator/ros-bridge #308: ImportError: cannot import name CarlaTrafficLightInfo

dmrly, 2020-06-07:
i have a issue `ImportError: cannot import name CarlaTrafficLightInfo` `Run roslaunch carla_ros_bridge carla_ros_bridge_with_rviz.launch`. 
```
export PYTHONPATH=$PYTHONPATH:/home/nd/carla/PythonAPI/carla/dist/carla-0.9.9-py2.7-linux-x86_64.egg 
 python -c 'import carla;print("Success")'
Success
export PYTHONPATH=$PYTHONPATH:/home/nd/carla/PythonAPI/carla/dist/carla-0.9.9-py3.6-linux-x86_64.egg
python -c 'import carla;print("Success")'
Success
```
Carla , Ros and UE4 works properly.

i got issue:
```
Traceback (most recent call last):
  File "/home/nd/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 33, in <module>
    from carla_ros_bridge.traffic import Traffic, TrafficLight
  File "/home/nd/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/traffic.py", line 15, in <module>
    from carla_msgs.msg import CarlaTrafficLightStatus, CarlaTrafficLightInfo
ImportError: cannot import name CarlaTrafficLightInfo
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 3749, exit code 1, cmd /home/nd/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/nd/.ros/log/d7a93a84-a8a8-11ea-9bf2-606c66522637/carla_ros_bridge-2.log].
log file: /home/nd/.ros/log/d7a93a84-a8a8-11ea-9bf2-606c66522637/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
```
What can I do about this topic?

NmBoyd, 2020-06-10:
I had this issue as well. The documentation [](https://carla.readthedocs.io/en/latest/ros_installation/) is different than the developer install on the github page. Namely it forgets the following to update the message submodule.
```
cd ros-bridge
git submodule update --init
```

dmrly, 2020-06-11:
@NmBoyd  I did the operation on the file you specified, but the same error continues? What else can I do?

NmBoyd, 2020-06-11:
Can you please verify that the following files are correct in your directory? Just run a simple ls in the carla_msg folder. 
```
<path_to_folder>/carla-ros-bridge/ros-bridge/carla_msgs/msg$ ls
CarlaActorInfo.msg            CarlaLaneInvasionEvent.msg
CarlaActorList.msg            CarlaStatus.msg
CarlaBoundingBox.msg          CarlaTrafficLightInfoList.msg
CarlaCollisionEvent.msg       CarlaTrafficLightInfo.msg
CarlaControl.msg              CarlaTrafficLightStatusList.msg
CarlaEgoVehicleControl.msg    CarlaTrafficLightStatus.msg
CarlaEgoVehicleInfo.msg       CarlaWalkerControl.msg
CarlaEgoVehicleInfoWheel.msg  CarlaWeatherParameters.msg
CarlaEgoVehicleStatus.msg     CarlaWorldInfo.msg
```
if the files either 1) aren't there or 2) are missing CarlaTrafficLightInfo then the submodule isn't initialized and updated properly.

dmrly, 2020-06-11:
Thank you for helping @NmBoyd ,
The carla_msg files were all, but not working, I deleted and re-installed. And now I'm getting this error: 
`Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/nd/carla/PythonAPI/carla/dist/carla-0.9.9-py3.6-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "/home/nd/carla/PythonAPI/carla/dist/carla-0.9.9-py3.6-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "/home/nd/carla/PythonAPI/carla/dist/carla-0.9.9-py3.6-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
ImportError: /home/nd/.cache/Python-Eggs/carla-0.9.9-py3.6-linux-x86_64.egg-tmp/carla/libcarla.cpython-36m-x86_64-linux-gnu.so: undefined symbol: PyUnicode_AsUTF8String
`
After installing the developer correctly, from here:  https://github.com/carla-simulator/ros-bridge
run `export PYTHONPATH=$PYTHONPATH:/home/nd/carla/PythonAPI/carla/dist/carla-0.9.9-py3.6-linux-x86_64.egg`
and then run `python3 -c 'import carla;print("Success")'`   i got `Succes`

run `export PYTHONPATH=$PYTHONPATH:/home/nd/carla/PythonAPI/carla/dist/carla-0.9.9-py2.7-linux-x86_64.egg`
then `python -c 'import carla;print("Success")'`
i got error.

What else can I try for a solution?

dmrly, 2020-06-11:
and now i get error : `[FATAL] [1591898208.421048, 0.000000]: CARLA python module version 0.9.9 required. Found: 0.9.5
================================================================================
REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 3205, exit code 1, cmd /home/nd/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/nd/.ros/log/e8cc20b2-ac0c-11ea-beca-606c66522637/carla_ros_bridge-2.log].
log file: /home/nd/.ros/log/e8cc20b2-ac0c-11ea-beca-606c66522637/carla_ros_bridge-2*.log
Initiating shutdown!
`
run `roslaunch carla_ros_bridge carla_ros_bridge.launch`

fpasch, 2020-06-15:
As the error states: Please use the corresponding CARLA version 0.9.9 for server and python egg (in PYTHONPATH)

aldyrenaldy00, 2020-06-16:
Hello, i also happened to have this issue, the CarlaTrafficLightInfo error still occur after 
`git submodule update --init` 
I also already checked the msg files in the directory and replaced it with  [msgs](https://github.com/carla-simulator/ros-carla-msgs/tree/14d8af014f4517a118390a9666dafa87c80169fc)

but the error still not fixed,  theni changed the folder name ros-bridge in catkin_ws/src/ into ~ros-bridge, and the CarlaTrafficLightInfo is fixed, but i get another error

Traceback (most recent call last):
  File "/home/aldyrenaldy00/carla-ros-bridge/catkin_ws/src/~ros-bridge/~carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 26, in <module>
    from carla_ros_bridge.actor import Actor
ImportError: No module named actor

thank you

faultfactory, 2020-06-24:
After you do the submodule init you may need to rebuild. Adding the repository does not necessarily add the python files, just the msg files.

joel-mb, 2020-07-17:
Hi @dmrly @aldyrenaldy00 , are you still facing the issue?
As @NmBoyd @faultfactory said, you should init/update the messages submodule (`git submodule update --init`) and then run the `catkin_make` command again (all the commands are specified here: https://github.com/carla-simulator/ros-bridge#for-developers). @NmBoyd Thank you for pointing out the missing information from the CARLA documentation!

On the other hand, this issue is not yet solved if you are installing the ros-bridge using the debian package (i.e., via `apt-get install`). We are waiting to get the `carla-msgs` package updated in the ROS index.

joel-mb, 2020-08-31:
The `carla_msgs` package has been updated in the ROS index, so now all the messages are also available when installing the `ros-bridge` using the debian package. Feel free to re-open the issue if needed.

closed 2020-08-31 by joel-mb

luoxinkaikevin, 2022-04-27:
i also meet this problem, did you solve it @Nurullahdmrly @joel-mb @aldyrenaldy00 @NmBoyd 
thanks for your attention!

Anhar2, 2024-03-18:
> i also meet this problem, did you solve it @Nurullahdmrly @joel-mb @aldyrenaldy00 @NmBoyd thanks for your attention!

could you solve it? Anyone?