carla-simulator/ros-bridge #310: Ros-bridge: Version mismatch detected: Client API and Simulator API

MichaelKlin, 2020-06-08:
I installed ros-bridge, but i get this error: 
[INFO] [1591552433.559553, 0.000000]: Waiting for CARLA world (topic: /carla/world_info)...
[INFO] [1591552433.852305, 0.000000]: Trying to connect to 10.0.2.2:2000
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 4dc4cb81 
WARNING: Simulator API version  = 0.9.9-1-g67684c5b 
terminate called after throwing an instance of 'std::bad_alloc'
what():  std::bad_alloc
I recompiled my PythonAPI without errors, but Client API version is the same= 4dc4cb81.
If i delete all egg-files, i get also the same error. I do not know, what i have to do. 
My eggs are in home/users/Dokuments/...  and my path looks like PYTHONPATH=$PYTHONPATH:./Documents/carla/PythonAPI/carla/dist/carla-0.9.9-py2.7-linux-x86_64.egg
My steps are:
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 81061A1A042F527D &&
sudo add-apt-repository "deb [arch=amd64 trusted=yes] http://dist.carla.org/carla-ros-bridge-melodic/ bionic main"
mkdir -p ~/carla-ros-bridge/catkin_ws/src
cd ~/carla-ros-bridge
git clone https://github.com/carla-simulator/ros-bridge.git
cd ros-bridge
git submodule update --init
cd ../catkin_ws/src
ln -s ../../ros-bridge
source /opt/ros/melodic/setup.bash
cd ..
rosdep update
rosdep install --from-paths src --ignore-src -r
PYTHONPATH=$PYTHONPATH:./Documents/carla/PythonAPI/carla/dist/carla-0.9.9-py2.7-linux-x86_64.egg
source ~/carla-ros-bridge/catkin_ws/devel/setup.bash       roslaunch carla_ros_bridge carla_ros_bridge.launch

fpasch, 2020-06-15:
Please check the order in your PYTHONPATH.

closed 2020-07-15 by fpasch

MichaelKlin, 2020-09-10:
@fpasch Thanks u, i get it