carla-simulator/ros-bridge #311: Sensor initialization issue in CARLA ego-vehicle node

rohanb2018, 2020-06-09:
In the CARLA ego-vehicle [node](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ego_vehicle/src/carla_ego_vehicle/carla_ego_vehicle.py#L177), there seems to be a bug with initializing certain sensors in the `setup_sensors` method. 

In particular, for sensors that are not explicitly handled (such as the collision and lane invasion sensors), it is possible that `sensor_location` and `sensor_rotation` can either (a) fail to be initialized at all (if the sensor is the first sensor in the JSON config file), or (b) will have incorrect values corresponding to the previously-handled sensor.

My solution was to add an `else` clause to handle all sensors that are not explicitly handled.

```
else:
    sensor_location = carla.Location(x=sensor_spec['x'], y=sensor_spec['y'], z=sensor_spec['z'])
    sensor_rotation = carla.Rotation(pitch=0.0, roll=0.0, yaw=0.0)
```

fpasch, 2020-06-15:
Sounds reasonable. Would you be able to submit a PR?

rohanb2018, 2020-06-17:
@fpasch Sure! I will submit a PR as soon as I can and link it to this issue.

closed 2020-10-26 by fpasch