carla-simulator/ros-bridge #316: Carla_ad_demo Scenario

Roosstef, 2020-06-15:
Hi all,

is there a way to run other scenarios from scenario runner for example FollowLeadingVehicle.xml?
When I try to run FollowLeadingVehicle.xml within the carla_ad_demo.launch the following error occures:

[image]

fpasch, 2020-06-15:
That's possible. But only open scenarios are supported.
Please be aware, that we're currently reworking the scenario execution in #309

Roosstef, 2020-06-15:
thanks for the fast respond!
Currently only the FollowLeadingVehicle.xosc included in the ros_bridge can be used right?
Other .xosc files from scenario_runner cant be used?

The Target Vehicle FollowLeadingVehicle.xosc included in the ros_bridge is not moving in my case. Am I doing something wrong?

fpasch, 2020-06-15:
You can of course use other openscenarios. Just include them here:
https://github.com/carla-simulator/ros-bridge/blob/master/carla_ad_demo/launch/carla_ad_demo_with_scenario.launch#L77

Regarding the vehicle not moving, please try it again with the latest master. We've updated the scenario.

Roosstef, 2020-06-15:
Ok thank you, but the default scenarios (.xosc) from open_scenario cant run out of the box with carla_ad_demo.launch right?
There are some parameters that need to be changed in the (.xosc) files like ScenarioObject are named hero, not ego_vehicle etc.?

fpasch, 2020-06-15:
These example scenarios might need some modifications:
- Locations (if you have the corresponding opendrive map, then you might be able to load it in CARLA and keep the locations)
- Names of the vehicles (which vehicle should be controlled by an AD stack?)

In general, there are two options to instantiate an AD stack together with a scenario:
1. Instantiate it by defining it in the scenario (see #318 for some documentation)
2. Instantiate it externally (then you might need to provide the route and target-speed yourself)

Roosstef, 2020-06-16:
I dont understand how to specifiy a detination for the ego_vehicle within the scenario file. For example when I try to run the CyclistCrossing.xosc the ego_vehicle follows a random (?) waypoint and is not driving in the direction where the cyclist is. By manually setting the waypoint next to the cyclist via rviz it is working but I want that to be handled automatically. Is there a way to?
Thanks for your help! Sry but I'm new to OpenScenario.

fpasch, 2020-06-17:
Setting the waypoints and the target speed is now possible within the openscenario. Example:

    <Action name="EgoVehicleRouteAction">
        <PrivateAction>
            <RoutingAction>
                <AssignRouteAction>
                    <Route name="EgoVehicleRoute" closed="false">
                        <Waypoint routeStrategy="fastest">
                            <Position>
                                <WorldPosition x="275" y="-173" z="1" h="0"/>
                            </Position>
                        </Waypoint>
                        <Waypoint routeStrategy="fastest">
                            <Position>
                                <WorldPosition x="250" y="-173" z="1" h="0"/>
                            </Position>
                        </Waypoint>
                    </Route>
                </AssignRouteAction>
            </RoutingAction>
        </PrivateAction>
    </Action>

closed 2020-07-15 by fpasch

Roosstef, 2020-07-15:
Thanks for your advice. 

I tried to use this example for FollowLeadingVehicle.xosc but it doesnt work for me.
Can you please give an example with FollowLeadingVehicle.xosc?

fpasch, 2020-07-15:
Unfortunately, that's out of our scope here. Please have a look at the [official documentation](https://releases.asam.net/OpenSCENARIO/1.0.0/Model-Documentation/index.html) (especially the "XSD 1.0 Tree View" helps to understand where to define specific xml-elements) or if you have specific questions regarding the execution within CARLA, please raise your question [here](https://github.com/carla-simulator/scenario_runner/issues)