carla-simulator/ros-bridge #321: Target Vehicle not moving in FollowLeadingVehicle scenario

dvh549, 2020-06-17:
Hi all,

I tried executing the scenario and waited for the target vehicle to move but it didn't. I tried multiple times but the scenario always fails at execution and status turns to red. However, the spawned NPCs moved as per normal. Upon closer look, I found these errors in the terminal as shown in the below images. Kindly advise on how to proceed.

Carla Version: 0.9.9.2
OS: Ubuntu 18.04

[image]
[image]

fpasch, 2020-07-15:
Are you still able to reproduce the issue?

dvh549, 2020-07-15:
Hi @fpasch , I do not face this error anymore. Thanks.

closed 2020-07-15 by fpasch