carla-simulator/ros-bridge #322: Ros noetic bridge

Roosstef, 2020-06-17:
Hey, are there any official plans to make the ros_bridge compatible With ubuntu 20.04 to use ist With the new Python3 Ros Distro?

hofbi, 2020-06-17:
I am currently playing around with this and will submit a draft PR to track the status. So far there are some missing packages such as `derived-object-msgs`

togaen, 2020-07-15:
I have also played around with this on Ubuntu 20.04 with ROS Noetic, and the current master of the ros bridge seems to work well, at least for basic functionality. I only ran into three minor issues:

1. `pcl_recorder`: I needed to build this with C++14 enabled
1. `carla_ros_bridge`: I needed to change `self.actors.iteritems()` to `self.actors.items()` on [this line](https://github.com/carla-simulator/ros-bridge/blob/1bd54a3814e321d8ef072cc299aed90079d0b536/carla_ros_bridge/src/carla_ros_bridge/bridge.py#L229)
1. `rviz_carla_plugin`: this package did not build, but I don't need it, so I added CATKIN_IGNORE

hofbi, 2020-07-15:
Thanks for the update. I added you feedback to my draft PR #323. Currently I am just waiting for the release of the python catkin tools https://github.com/catkin/catkin_tools/issues/594

closed 2021-03-26 by joel-mb