carla-simulator/ros-bridge #327: cannot import name CarlaTrafficLightInfo

huangxueqin123, 2020-06-29:
I am having a similar problem with https://github.com/carla-simulator/ros-bridge/issues/263#issue-604840570.

Installing the most recent package

### sudo apt install carla-ros-bridge-kinetic

Then running
### roslaunch carla_ros_bridge carla_ros_bridge.launch

I get an error that it cannot import CarlaTrafficLightInfo. 
I found that there is no carla_msgs folder. Then I manually cloned and put it in  /opt/carla-ros-bridge/kinetic/lib/python2.7/dist-packages/carla_ros_bridge. But i didn't know how to rebuild it.

konradst, 2020-07-18:
@huangxueqin123 https://carla.readthedocs.io/en/latest/ros_installation/

joel-mb, 2020-08-31:
Hi @huangxueqin123 ,
All the debian packages have been finally updated, so the issue should have disappeared. Follow the next instructions to install the new packages:

```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9
sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"

sudo apt-get update
sudo apt-get install carla-simulator
sudo apt-get install carla-ros-bridge
```

I would recommend removing all the entries pointing to CARLA apt repositories from `/etc/apt/sources.list` to avoid any conflict. Please, note that the instructions above will install the latest version of CARLA (i.e., pre-release 0.9.10).

Feel free to re-open the issue if needed.

closed 2020-08-31 by joel-mb