carla-simulator/ros-bridge #331: Subscribers missing first collision event

GomesJM, 2020-07-14:
The `Communication::publish_message()` function creates a `Publisher` object once the first collision event occurs and immediately publishes to the topic. Subsequent events then publish to the existing publisher.

The issue is that subscribers do not receive the first event. Perhaps due to pub-sub connections not being established in time?

The same is true for lane invasion events and likely all non-latching messages published through this function.

For now, I can work around this issue by modifying the code and setting the `is_latched` flag for the collision and lane invasion topics.

However, it seems like a possibly "better" solution might be to initialize the `Publisher` object as soon as a sensor is created, so that the publisher is ready once an event/msg is received.

joel-mb, 2020-07-17:
Hi @GomesJM! Thank you for the information.
I'll try to replicate the issue. It makes sense to me to initialize the `Publisher` object as soon as a sensor is created.

fpasch, 2020-07-17:
This will change with the new ROS/ROS2 "hybrid" bridge.
You can find the current state [here](https://github.com/carla-simulator/ros-bridge/tree/ros2).

Could you please test it with that version?

closed 2020-08-17 by GomesJM