carla-simulator/ros-bridge #332: [Question] Map visualization on RVIZ

bbrito, 2020-07-14:
Dear ros-bridge team,

Is it possible to visualize the map of the city in RVIZ?

Best regards,
Bruno

joel-mb, 2020-07-17:
Hi @bbrito!
Unfortunately, we don't have any package to visualize the CARLA towns out of the box yet. So far, I can think of two possibilities:

* Using the point cloud maps provided in this repo (https://bitbucket.org/carla-simulator/autoware-contents.git). We are using this approach in the integration with autoware.
* Try to export the mesh from Unreal and then use a ros Marker to visualize it.

Hope this helps!

bbrito, 2020-07-17:
Hi @joel-mb!

Thanks for the advice. I will try the first option.

Best regards,
Bruno

closed 2020-07-17 by bbrito

konradst, 2021-03-18:
Why have you closed it?