carla-simulator/ros-bridge #335: ValueError: cannot reshape array, lidar misaligned

konradst, 2020-07-17:
Hello,
I'm trying to `roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch` and get spammed by sensor error and lidar goes out of order (it detects a wall but then show it spread into 4 chunks all around the vehicle). Camera and odometer work fine. I've tried the following setup:
Ubuntu Bionic
Python 2.7
Unreal: **4.24** (from source)
Carla: **0.9.9.4, 0.9.9** (from source)
ROS: **Melodic** (from apt)
ros bridge: **master** (from source)

I use rviz so I haven't been able to use noetic or python3 yet to even test if it would work properly.

The errors I get:
```
ValueError: cannot reshape array of size 46053 into shape (11513,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46050 into shape (11512,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46074 into shape (11518,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46029 into shape (11507,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46038 into shape (11509,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46011 into shape (11502,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46011 into shape (11502,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 45969 into shape (11492,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 45969 into shape (11492,4)
^CTraceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 45969 into shape (11492,4)
```

fpasch, 2020-07-22:
This is most likely related to CARLA 0.9.9.4 not containing https://github.com/carla-simulator/carla/pull/2997
Please test with the latest master.

closed 2020-07-29 by fpasch

abhi5691, 2020-09-11:
Hello @konradst  could you solve this?

My configuration is precisely same as @konradst and I also incorporated the changes in PR#2997 shared by @fpasch , rebuilt the simulator, but got the same blast of errors as shared above

I also tried to follow @fpasch 's suggestion of using current master but then I no longer get egg file for Python2, which is necessary for me as I am using ros-bridge. Can someone suggest what can be at the root of this issue?

konradst, 2020-09-11:
@abhi5691 I only suppressed the error with 2>/dev/null for the console's relief ;) and forgot about the issue. I'm waiting for ros noetic bridge to happen.

fpasch, 2020-09-21:
@konradst noetic bridge was just merged.