carla-simulator/ros-bridge #338: /tf_static not published

alex75016, 2020-07-24:
Hello, I'm currently using Carla and carla-ros-bridge to test a SLAM algorithm (Cartographer). To do so, I record a bagfile containing the lidar, imu and tf messages. The problem is that I need to record the /tf_static which appear on the rostopic list when Carla and carla_ego_vehicle are active, but when I do rostopic info /tf_static, it shows that there is no publisher of /tf_static and only a subscriber to it. 
I was then wondering if it is normal and if not, is there a way to get /tf_static published ?

Thank you

fpasch, 2020-07-27:
What transforms are you expecting?

alex75016, 2020-07-27:
I'm expecting the transforms between the lidar frame and the vehicle frame, and also between the IMU frame and the vehicle frame.

alex75016, 2020-07-27:
> What transforms are you expecting?

I'm expecting the transforms between the lidar frame and the vehicle frame, and also between the IMU frame and the vehicle frame.