carla-simulator/ros-bridge #344: Python 3 Support

hishamyousry, 2020-08-10:
I am using the ros-bridge with CARLA 0.9.9 and Python 3.6. When running this example:
`roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch`
I get this error multiple times:

`Traceback (most recent call last):`
`  File "/opt/ros/melodic/lib/python3/dist-packages/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data`
`    self.sensor_data_updated(carla_sensor_data)`
`  File "/opt/ros/melodic/lib/python3/dist-packages/carla_ros_bridge/lidar.py", line 64, in sensor_data_updated`
`    carla_lidar_measurement.raw_data, dtype=numpy.float32)`
`TypeError: fromstring() argument 1 must be read-only bytes-like object, not memoryview`

I googled a bit and found that the way fromstring() handles data is different in Python 3 than Python 2, so this is probably why I am getting this error. 
So my question is this: Is Python 3 even supported? I know that CARLA supports both Python 2 and 3, but is this the case with the ROS bridge? If it should be supported, then this should probably be handled.

P.S. I see that fromstring() was recenty added in #330. Not sure if Python 3 was compatible before that.

hofbi, 2020-08-10:
I am currently working on supporting ROS Noetic which uses Python 3 in #323. With this PR all packages will support Python 3.

hishamyousry, 2020-08-13:
@hofbi Thanks. In case it helps, I just confirmed that the version of lidar.py commited on June 15th, 2020 (d0437c12f51ede053c9f637dc02dd38d0e1ebec6) is working fine with Python 3.

closed 2020-09-30 by fpasch