carla-simulator/ros-bridge #345: Lauching the Ros bridge

Karthikaddagadderamesh, 2020-08-10:
Hello all,

When i trying to run the command: 'roslaunch carla_ros_bridge carla_ros_bridge.launch' 

I am gettingthe following error: ImportError: /home/../.cache/Python-Eggs/carla-0.9.7-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so: undefined symbol: PyUnicode_AsUTF8String
REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 19218, exit code 1

I have exported the and tried both 2.7 and 3.5 .egg
still the issue exists

hofbi, 2020-08-10:
Look like a duplicate of #38. So could you verify that you only use the pyhton 2 egg in your `PYTHONPATH`

Karthikaddagadderamesh, 2020-08-10:
I am usin
Ubantu: 18.04
python: 2.7 or 3.6
carla: 0.9.7
Initially i tried with with python 2 egg.
I got the following error '' CARLA python module version 0.9.9 required. Found: 0.9.7''

so I tried python3 egg

hofbi, 2020-08-10:
The default python version for ROS melodic is Python 2, so you should use the python 2 eggs. Please make sure that there is only one carla egg in your PYTHONPATH

closed 2020-08-11 by Karthikaddagadderamesh