carla-simulator/ros-bridge #348: Is semi-truck available?

manonechu, 2020-08-14:
Hi, thank you for the great simulator.

I would like to know that is a semi-truck available? Because I'd like to simulate autonomous driving for a semi-truck. If it is not available, is there a way to create or import it? Please kindly advise me.

Thank you so much for your help!

manonechu, 2020-08-14:
I would like to ask one more question on the waypoints for a vehicle to follow: how should I publish select these waypoints on the map and then pass them to the vehicle to follow? Is there a tool available to do that?
By the way, I would use roscpp to develop nodes.
Thank you again!

fpasch, 2020-08-14:
Please ask your question regarding additional vehicles [here](https://github.com/carla-simulator/carla).

Regarding the routing, please have a look [here](https://github.com/carla-simulator/ros-bridge/tree/master/carla_waypoint_publisher). For gathering the positions I would recommend using manual control to navigate through the town.

closed 2020-08-18 by fpasch