carla-simulator/ros-bridge #355: CARLA python module - FATAL

TNAV1, 2020-08-20:
New to CARLA.  Ubuntu 16.04, ROS Kinetic, Carla0.9.8.

Trying to get the carla-ros-bridge running while running a server launch per...
https://github.com/carla-simulator/ros-bridge (Start the ROS bridge).

The server launch shows Town03.
However, 
roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch
results in the following error 

**Trying to connect to localhost:2000
CARLA python module version 0.9.9 required. Found: 0.9.8
signal_shutdown [atexit]**

I previously started to install CARLA0.9.9 but deleted it and started again with 0.9.8.  There should be no files left pointing to 9.9.  
What am I missing?

TNAV1, 2020-08-20:
Is 0.9.9 hard coded in carla-rod-bridge. Do I need to pull an older version and reinstall?

fpasch, 2020-08-21:
Yes, the ROS bridge is depending on a specific version of CARLA, due to possible changes within the Python API.

closed 2020-08-21 by fpasch