carla-simulator/ros-bridge #359: Carla AD demo bloom_intensity error

Roosstef, 2020-08-25:
Hey all,

today I updated carla (0.9.9.9.4-106) and ros-bridge. When trying to launch carla_ad_demo.launch the package crashes with following error:

[image]

Before the update I had no such errors.
Maybe something changed on Camera parameters?
I hope someone can help me about this issue.

cmpute, 2020-08-25:
I also met this problem. First I added the definition of `bloom_intensity` in the sensor definition file, afterwards I got another error
```
File ".../carla_ego_vehicle/src/carla_ego_vehicle/carla_ego_vehicle.py", line 262, in setup_sensors
    bp.set_attribute('bloom_intensity', str(sensor_spec['bloom_intensity']))
IndexError: attribute 'bloom_intensity' not found
```

Roosstef, 2020-08-31:
Any ideas how to fix this problem? I have not found a solution yet.

leodario7, 2020-09-07:
Same problem over here, it seems to be a new definition not present in version 0.9.9, I have not been able to try it out yet

Roosstef, 2020-09-08:
Seems to be fixed in newest update.

closed 2020-09-08 by Roosstef