carla-simulator/ros-bridge #366: LiDAR data ValueError: cannot reshape array of size X into shape Y

abhi5691, 2020-09-11:
Precisely as described in #335 , I face issue with LiDAR data via carla-ros-bridge
My configuration is exactly same as @konradst:
Ubuntu Bionic
Python 2.7
Unreal: 4.24 (branch from source)
Carla: 0.9.9.4 (tag from source)
ROS: Melodic (from apt)
ros bridge: master (branch from source)
pygame installed with pip

I also incorporated the changes in PR#2997, rebuilt the simulator, but got the same blast of errors as in original issue (#335)

I tried to follow @fpasch 's suggestion on #335 of using current master but then I no longer get egg file for Python2, which is necessary for me as I am using ros-bridge: so I haven't been able to use python3 yet to even test if it would work properly. Can someone suggest what can be at the root of this issue?

The errors:

```
ValueError: cannot reshape array of size 46053 into shape (11513,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46050 into shape (11512,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46074 into shape (11518,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46029 into shape (11507,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46038 into shape (11509,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46011 into shape (11502,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 46011 into shape (11502,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 45969 into shape (11492,4)
Traceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 45969 into shape (11492,4)
^CTraceback (most recent call last):
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
    self.sensor_data_updated(carla_sensor_data)
  File "/home/konrad/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
    lidar_data, (int(lidar_data.shape[0] / 4), 4))
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 45969 into shape (11492,4)
```

joel-mb, 2020-09-14:
Hi @abhi5691 !
As you have noticed, the egg file for Python2 is not automatically generated. To generate it just use the following command:
```sh
make PythonAPI ARGS="--python-version=2"
```
This will allow you to use the latest master and therefore incorporate https://github.com/carla-simulator/carla/pull/2997.
Hope this solves the issue!

sulichuan, 2021-03-11:
Hi
I have the same issue when I played "roslaunch carla_ego_vehicle carla_example_ego_vehicle.launch"
and I had generate the egg file with "make PythonAPI ARGS="--python-version=2""

the error message is still exist

maximusron, 2021-05-14:
> Hi @abhi5691 !
> As you have noticed, the egg file for Python2 is not automatically generated. To generate it just use the following command:
> 
> ```shell
> make PythonAPI ARGS="--python-version=2"
> ```
> 
> This will allow you to use the latest master and therefore incorporate [carla-simulator/carla#2997](https://github.com/carla-simulator/carla/pull/2997).
> Hope this solves the issue!

Hi can you please walk me through the fix and where to run this command

te77yhou, 2021-09-23:
Just add a line `lidar_data = lidar_data[: len(lidar_data) - len(lidar_data) % 3]` before numpy.reshape(line 82) in `/opt/carla-ros-bridge/melodic/lib/python2.7/dist-packages/carla_ros_bridge/lidar.py` fix this "issue"