carla-simulator/ros-bridge #368: Import Error: undefined symbol: _PyThreadState_Current

KilinJin, 2020-09-13:
Hi, I'm trying to run ros-bridge following the guide (https://github.com/carla-simulator/ros-bridge#start-the-ros-bridge), but I'm having the following error when starting the ros-bridge using option 1 (roslaunch carla_ros_bridge carla_ros_bridge.launch):
'''
... logging to /home/qilin/.ros/log/a4187af0-f573-11ea-ba27-50eb71236e19/roslaunch-qilin-Alienware-Aurora-R9-17727.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://qilin-Alienware-Aurora-R9:44775/

SUMMARY
========

PARAMETERS
 * /carla/ego_vehicle/role_name: ['hero', 'ego_veh...
 * /carla/fixed_delta_seconds: 0.05
 * /carla/host: localhost
 * /carla/port: 2000
 * /carla/synchronous_mode: False
 * /carla/synchronous_mode_wait_for_vehicle_control_command: True
 * /carla/timeout: 10
 * /rosbag_fname: 
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /use_sim_time: True

NODES
  /
    carla_ros_bridge (carla_ros_bridge/bridge.py)

auto-starting new master
process[master]: started with pid [17737]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a4187af0-f573-11ea-ba27-50eb71236e19
process[rosout-1]: started with pid [17748]
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [17752]
ModuleSpec(name='carla.libcarla', loader=<_frozen_importlib_external.ExtensionFileLoader object at 0x7f840dc94670>, origin='/home/qilin/.cache/Python-Eggs/carla-0.9.9-py2.7-linux-x86_64.egg-tmp/carla/libcarla.so')
Traceback (most recent call last):
  File "/opt/carla-ros-bridge/melodic/lib/carla_ros_bridge/bridge.py", line 24, in <module>
    import carla
  File "<frozen zipimport>", line 259, in load_module
  File "/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.9-py2.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.9-py2.7-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.9-py2.7-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
  File "/home/qilin/anaconda3/lib/python3.8/imp.py", line 343, in load_dynamic
    print(_load(spec))
ImportError: /home/qilin/.cache/Python-Eggs/carla-0.9.9-py2.7-linux-x86_64.egg-tmp/carla/libcarla.so: undefined symbol: _PyThreadState_Current
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 17752, exit code 1, cmd /opt/carla-ros-bridge/melodic/lib/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/qilin/.ros/log/a4187af0-f573-11ea-ba27-50eb71236e19/carla_ros_bridge-2.log].
log file: /home/qilin/.ros/log/a4187af0-f573-11ea-ba27-50eb71236e19/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
[carla_ros_bridge-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
'''

It seems to be an import carla problem, but the wired thing is, I can import carla correctly when writing my own code, also using python 2.7.

My Carla works well and I installed my ROS was built following the guide on their official website. 
I'm using Ubuntu 18.04.

Also an additional question, what is the recommended python version for running ros-bridge for carla? 

Thank you!

joel-mb, 2020-09-14:
Hi @KilinJin !
As you are using ROS melodic, you should always use Python2.7. 
It seems that you are trying to import the egg files for Python2 using Python3.  Make sure that you use Python2 and add also to the `PYTHONPATH` the egg file for Python2.7

I hope this helps!

KilinJin, 2020-09-14:
Hi @joel-mb !
Thank you for the reply, I think I did exactly what you said, here are my settings:
bashrc:
[image]
python version:
[image]
python console:
[image]

Could you please tell me if there's anything wrong?

Thank you!

KilinJin, 2020-09-14:
Hi I think I've fixed the problem, there's some python version issue in my computer because of Anaconda. Everything works fine for me now.

Thanks!

closed 2020-09-14 by KilinJin