carla-simulator/ros-bridge #375: /carla/objects does not publish anything

Cram3r95, 2020-09-24:
Hi,

I am using CARLA 0.9.9.4 (tag, June 18 commit) and ros-bridge (May 26). I am trying to obtain the content of /carla/objects topics. The code seems to be perfect (traffic_participant.py sends the message to pseudo_actor, and then pseudo_actor to communication.py). However, when I try to obtain the content of this topic, it does not seem to be publishing anything. On the other hand, I have appreciated another topic ("carla/ego_vehicle/objects"), that again does not publish anything.

Moreover, I tried to pull and compile the latest version of the carla-ros-bridge (which fixes the classification certainty value) but it seems to be incompatible with my version of CARLA:

[image]

What can I do?

joel-mb, 2020-10-19:
Hi @Cram3r95!
Sorry for the late answer. It took me a while to figure out what was happening.

As I told you in discord, I was having exactly the same issue and seems to be related to some kind of mismatch between some ROS debian packages. Actually, I was able to reproduce the problem using only `rostopic pub` and `rostopic echo`, so the CARLA ROS bridge is not producing the issue (can you check this on your machine?)

Specifically, I was able to reproduce the issue with the `ros-melodic-ackermann-msgs` and `ros-melodic-derived-msgs` packages. For some reason, if you have the latest version of those packages installed with a prior installation of ROS melodic, it doesn't work. Whereas,  with a fresh installation of ros-melodic everything works fine.

As a workaround, I'm using a snapshot of the ROS apt repository (http://wiki.ros.org/SnapshotRepository) to install an older version of the required packages. You can take a look of how I'm using the snapshot here: https://github.com/carla-simulator/carla-autoware/blob/master/Dockerfile#L22

Hope this helps!

Cram3r95, 2020-10-26:
@joel-mb sorry for the late answer, I did not check that problem before since now we are focused on CARLA challenge. Today I conducted a fresh installation of ros-melodic and derived/ackermann messages and it seems to be perfect! If I find any new problem related to this issue, I will comment in a new issue.

closed 2020-10-26 by Cram3r95