carla-simulator/ros-bridge #398: Simulation time slows down in synchronous mode

francismaria, 2020-10-21:
Hi there,

I am opening this issue because I am trying to make carla ROS bridge work in synchronous mode but I haven't been able to, despite my efforts. I need to collect the synchronous data from the ROS topics so that I can save it for later processing (using `rosbag`).

-------------

#### System Information:

- Server: Remote machine with NVIDIA RTX 2070 SUPER

- Client: Virtual machine w/ Ubuntu 18.04

#### System information:

- Carla Version: 0.9.9
- ROS Bridge version: 0.9.8
- Carla Server: docker running on the remote machine. 

```
docker run -p 2000-2002:2000-2002 --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=0 carlasim/carla:0.9.9 /bin/bash CarlaUE4.sh -benchmark -fps=20
```
-----------

#### Issue

When I try to run carla ROS Bridge with the following command:

```
roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch synchronous_mode:=true fixed_delta_seconds:=0.05
```

I notice that the simulation time is completely slowed down comparing to when in the asynchronous mode. Basically, a second in the simulation time is equal to 20 seconds in real-time. So it is in fact running 20 FPS of the simulation time, which corresponds to 20 seconds in real-time (1 FPS of the real-time).

I might be missing something really simple but I have not been able to figure out what's wrong. If someone could help me I would really appreciate it!

Thanks in advance!

fpasch, 2020-10-23:
Please ensure that ros bridge is not waiting for a vehicle command.
```
synchronous_mode_wait_for_vehicle_control_command:=false
```

francismaria, 2020-10-24:
That was exactly it! Thank you!

closed 2020-10-24 by francismaria

FHof, 2020-11-16:
The [documentation about `synchronous_mode_wait_for_vehicle_control_command`](https://github.com/carla-simulator/ros-bridge/blob/9d979818b226da2f60586e42be918a34c49d4910/carla_ros_bridge/config/settings.yaml#L10) shows a question mark.
What exactly is this setting doing?
Should I enable this only when I start a carla_manual_control?

fpasch, 2020-11-16:
If `synchronous_mode_wait_for_vehicle_control_command` is True and synchronous mode is enabled, ros-bridge waits for a vehicle-control command before sending the tick to CARLA.

Let's say you have a few sensors:
#### Example 1: synchronous_mode_wait_for_vehicle_control_command = False

Whenever sensor data of all expected sensors is received, the ros-bridge ticks (i.e. executes the next time step). No waiting for a vehicle control command.

#### Example 2: synchronous_mode_wait_for_vehicle_control_command = True

Whenever sensor data of all expected sensors is received __and__ the result of your AD stack (i.e. the vehicle control command) is received, the ros-bridge ticks.

FHof, 2020-11-18:
Thanks for the information. I record rosbag data for a replay of a state log running on the carla server, so I assume that there are no vehicle control commands.