carla-simulator/ros-bridge #406: Version mismatch detected:carla is incompatible with this API

changer707, 2020-10-28:
When I run CARLA AD demo,I got this problem as following:
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 4dc4cb81 
WARNING: Simulator API version  = 94cb1595 

No vehicle  is spawned in simulator.

joel-mb, 2020-11-02:
Hi @changer707!
As the WARNING says, you are using different versions of the CARLA client and the CARLA server. Sometimes this is not a problem and everything works as expected. Please, make sure you use the same version.
Are you getting any other error when running the CARLA AD demo?

dmorris0, 2021-01-02:
I am getting the same version mismatch problem, even though I think I'm doing the correct thing.  I am running Carla 0.9.10.1 binaries on my Windows machine, and I checked out ros-bridge with tag 0.9.10.1 and built this from source on Ubuntu 20.04 in WSL.  But when I run ros-bridge, I get the version warning and it fails to communicate with Carla:
```
[WARN] [1609390685.556875, 0.000000]: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API. Client API version: 784d9b9f. Simulator API version: 0.9.10.1
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version     = 784d9b9f
WARNING: Simulator API version  = 0.9.10.1
```
How can I build a ros-bridge client that is compatible with the simulator version 0.9.10.1?  I thought simply checking out the right tag would do it, but apparently not.