carla-simulator/ros-bridge #412: Over exposure in carla_manual_control camera feed afer 0.9.10.1 update

rowandempster, 2020-11-10:
Hi team,

I recently updating my research group's docker images to use the latest carla and carla-ros-bridge, after doing so and launching the carla_manual_control node as we normally do:
```
  <node pkg="carla_ego_vehicle" type="carla_ego_vehicle.py" name="carla_ego_vehicle" output="screen">
    <param name="sensor_definition_file" value="$(arg sensor_definition_file)" />
    <param name="vehicle_filter" value="$(arg vehicle_filter)" />
    <param name="spawn_point" value="$(arg spawn_point)" />
    <param name="spawn_ego_vehicle" value="$(arg spawn_ego_vehicle)" />
    <param name='role_name' value='wato_vehicle'/>
  </node>

  <include file="$(find carla_manual_control)/launch/carla_manual_control.launch" if="$(eval ego_gui)">
    <arg name='role_name' value='wato_vehicle'/>
  </include>
```
The manual control GUI is quite over exposed: https://cdn.discordapp.com/attachments/487687043745316864/775514074703986718/unknown.png 

I'm pretty sure that the 3rd person camera view is set up by the manual control node right? So our specific sensor configs shouldn't be messing it up?

Let me know what you think! Thanks :)

joel-mb, 2020-11-10:
Hi @rowandempster!
yeah, I think that your sensor configuration file maybe is messing it up. The manual control node is subscribed to the `view` camera defined in the sensor configuration file (https://github.com/carla-simulator/ros-bridge/blob/master/carla_ego_vehicle/config/sensors.json#L51 and https://github.com/carla-simulator/ros-bridge/blob/master/carla_manual_control/src/carla_manual_control/carla_manual_control.py#L88).

The default RGB parameters were updated in CARLA for the last release, so you should probably update these parameters in your configuration. Take a look here to know which parameters you need to change: https://github.com/carla-simulator/ros-bridge/commit/ee524bb50a7328402b4fe537a7e92d7927020bf8 As an alternative, if you want to use the default values, you can just remove the parameter from the sensor configuration file.

Hope this helps!

rowandempster, 2020-11-11:
Thanks for pointing that out @joel-mb, all set now :)

closed 2020-11-11 by rowandempster

parth3153, 2021-02-26:
hello, i have the same issue, and as per the instruction above i tried to change the parameter but no results, i am super confused the sensor.jason file is in carla_ego_vehicle/config and carla_autoware-agent/agent/config and i did changes in both.. 

from the Camera underexpose under windows, overexposed in Linux in v0.9.10.1 #3634  in https://github.com/carla-simulator/carla/issues/3634 there is success there aswell. 

its unclear how to make changes =, and in what files (are these 2 files which i mentioned)  and btw my carla version is the 0.9.9