carla-simulator/ros-bridge #415: Ubuntu 20.04 (Focal) No release file

EsquilliDev, 2020-11-16:
When running the code:

> sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9
>  sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"

I get the following error:

> E: The repository 'http://dist.carla.org/carla focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Any plans to fix that? I know we can use Melodic in the meantime, but I would like to get it working with Noetic.

Thanks in advanced!

hofbi, 2020-11-17:
We are currently working on it. Meanwhile you could follow the [developer setup](https://github.com/carla-simulator/ros-bridge#for-developers) to use the ros bridge on noetic.

hofbi, 2020-11-17:
Related issue in the carla simulator https://github.com/carla-simulator/carla/issues/3441

EsquilliDev, 2020-11-25:
Forgot to say that your suggestion actually worked!