carla-simulator/ros-bridge #416: Generating ring and time fields for lidars

Opletts, 2020-11-24:
Velodyne lidars publish `PointCloud2` messages with the `fields` array containing :
- `x` , type : `float32`
- `y` , type : `float32`
- `z` , type : `float32`
- `intensity` , type : `float32`
- `ring` , type : `uint16`
- `time` , type : `float32`

`ring` : Tells us which channel of the lidar the point belongs to.
`time` : Tells us the relative time at which the point was received during one rotation. If the `rotation_frequency` is **10Hz**, values would vary between **0-0.1s**)

Carla only publishes `XYZI` points, not `XYZIRT`. I think adding these two fields would be quite useful. I've made changes locally to generate `ring` in [lidar.py](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/lidar.py#L53) and can open a PR for review.

Generating the `time` field on the other hand is quite difficult. We've already received the binary data in [lidar.py](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/lidar.py#L53) so it's impossible to get that field here. I'm guessing changes to the lidar api in Carla would be required to make this possible. Any suggestions?

joel-mb, 2020-12-04:
Hi @Opletts!
Please, feel free to open a PR to add the `ring` field.
Regarding the `time` field, you are right. Modifications in CARLA would be required. I suggest opening an issue in the CARLA repo or contribute with your changes there.

Jaykumaran, 2025-08-27:
Hi @Opletts  @joel-mb ,

Was this featured added? I'm looking exactly for a solution of this kind to port a simulation from gazebo to carla.

Would be very helpful if i have this working.