carla-simulator/ros-bridge #422: Carla-Ros bridge and manula control in real-time

Lu7913, 2020-12-01:
Hi,
I would like to manually control vehicle in specific situations. The problem is, in order to do so, I need some decent fps count. When running Carla with Ros bridge I get maximum of 10 fps, with step size 0.1 and only one RGB camera spawned (wait for vehicle control is set to False). That is obviously to low to have some realistic results. 
I read that this is somewhat expected #185 

I'm nut sure if I'm doing something wrong, it is hardware issue or is this the way it is.
Before I loose more time over this I just want to check if it is possible to manually control vehicle in sort of real time via Carla-ROS bridge?

Thanks

--------------------
Kubuntu 18.04
Carla 9.10.1
Ros bridge 9.10.1

Quadro P4000 8 GB
Xeon silver 4114 2.2 GHz

joel-mb, 2020-12-04:
Hi @Lu7913 ,
With only one RGB camera should work much faster than that. Which is your environment? Are you just running the ROS bridge client in synchronous mode? Or do you have other CARLA clients?
I would run the following tests:
- Run the `manual control` script provided inside the CARLA project (inside `PythonAPI/examples`)
- Run the ROS bridge `manual control` with the default configuration (i.e., async mode).

Lu7913, 2020-12-04:
Hi, thank you for your replay.
ROS bridge is running in synchronous mode. No other Carla clients are running. For me, it doesn't make much difference in FPS count anyway. 
Running unmodified example `carla_ros_bridge_with_example_ego_vehicle`, results in even lover FPS count 3.5-6  in Town03

`manual control` inside Carla is running around 16 FPS in asynchronous mode, which is quite low as well.
Carla is built from source. It doesn't make any difference using Vulkan or OpenGL.

I have also tested fps count on prebuilt versions with manual control.py from Carla; on 0.9.10.1 I'm getting 17 fps in Town01 and Town03 20 fps on the server side.
With 0.9.9.4 I'm getting 25fps in Town03 and Town01.

Carla seems to be slow on its own, but should be so much slower with ROS bridge?

Lu7913, 2020-12-21:
The problem was underpowered GPU.

bbrito, 2021-01-25:
@Lu7913 which GPU are you using now? I am using a Titan V and it still does not run in real-time.

Lu7913, 2021-01-26:
Yes about that... 
So the first thing is that I fund out there are some problems with python2 and semantic camera, which is causing significant slow down is simulation, at least when running synchronous_mode.py. 
Then I was told that probably the reason is my GPU, which seems like it, when looking at GPU profiling in Unreal Editor. Then I found the issues below and I started looking at my CPU as well. 
#2968
#2689
Given how python is processed, my low single core performance, might also be the issue. I have also tried https://github.com/hofbi/telecarla, which is giving me some better results. I suspect that the reason is that python code is divided into two clients/two python scripts.

Looking at this all together I suspect that my problem is combination of  Carla architecture, bad single core performance CPU, and possibility that Carla need GPU for gaming? I'm using Quadro P4000 and I have also tried Quadro RTX 4000, which got me worse results and to me this is confusing. 
Using telecarla is giving me somewhat sufficient frame rate, for now, with the GPU I have and not to many sensors  spawned. 

I was planning on upgrading to better CPU (ryzen 9 5900x) and GPU (RTX 3080), but I'm not sure where is the main problem.

bbrito, 2021-01-26:
@Lu7913 thanks for the detailed answer. I have which I consider having a quite powerful GPU (Titan V) and CPU (Intel(R) Xeon(R) W-2123 CPU @ 3.60GHz) and still I get 3 to 5 FPS speed. I will try telecarla. Thanks for the tip!

But I think the problem is not hardware related. Is everyone getting this as the normal speed or am II doing something wrong to get such low performance?

Lu7913, 2021-01-26:
> But I think the problem is not hardware related. Is everyone getting this as the normal speed or am II doing something wrong to get such low performance?

I'm not sure also, but I'm out of other options. You saying that you dealing with similar issues, with better hardware, got me thinking again. But I really don't know what could I do wrong, when running unmodified examples.

bbrito, 2021-01-26:
@Lu7913 thanks for the detailed reply. I have which I consider a quite powerful GPU (Titan V) and CPU (Intel(R) Xeon(R) W-2123 CPU @ 3.60GHz) and I am getting really poor performance (3 to 5 FPS). I don't think this is a hardware issue. I will try telecarla. Thanks for the tip!

In the meantime, I have tried to use a variable simulation step (https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/#variable-time-step) and it runs almost realtime. Here they also suggest using a simulation step lower than 0.1 for synch simulations.

bbrito, 2021-03-18:
@Lu7913 I have tested this in Ubuntu 20 and Python3 and it is much faster. I think the main limitation for the real-time issues is actually using Python2. @joel-mb I think you can close this issue. Maybe add a warning in the documentation about this.

Lu7913, 2021-03-18:
Yes, on of the problems is Python2 (the most obvious is when running synchronous_mode.py, problem with semantic camera and Python2), the other thing is hardware. Few days ago I have tried Carla with new computer (RTX2070 and i7 9700k), and my frame rate is increased by 30-50% in some cases.

closed 2021-03-18 by Lu7913