carla-simulator/ros-bridge #438: ImportError: cannot import name 'is_within_distance_ahead' from 'misc'  when running the `carla_ad_demo`

bessszilard, 2020-12-15:
First of all, thank you for this useful open-source project. I am trying to run Carla ROS bridge demo launch file. 
I downloaded the catkin package as it [described](https://github.com/carla-simulator/ros-bridge#for-developers). These commands worked well:

```bash
# Option 1: start the ros bridge
roslaunch carla_ros_bridge carla_ros_bridge.launch

# Option 2: start the ros bridge together with an example ego vehicle
roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch
```

I exported the `SCENARIO_RUNNER_PATH` and I tried to run the [CARLA AD Demo](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ad_demo/README.md) with the command below, I receive the following errors [1]

```bash
roslaunch carla_ad_demo carla_ad_demo.launch
```

I tried to build the package with `catkin_make` and `catkin build` as well, but they behaved similarly. I tried to run this with the https://github.com/carla-simulator/ros-bridge/commit/537cc621404b10ad47b283918407306438a29f9f commit as well, but it behaved the same. I attached the full log as well [2]

What am I doing wrong?

Regards,
Szilard

Config:
|                 Config          |                                    |
| ---- | ---- |
| Operating system:         | Pop os 20.04 LTS                          |
| ROS version:              | noetic                                 |
| Carla simulation version: | CARLA_0.9.10.1 |
| ROS bridge version:       | [CARLA ROS Bridge 0.9.10.1](https://github.com/carla-simulator/ros-bridge/releases/tag/0.9.10.1) [`0700f2d`](https://github.com/carla-simulator/ros-bridge/commit/0700f2d24bf46b47188a03dce51615ee353d5aa9) |

[1] 
```bash
Traceback (most recent call last):
  File "/home/szilard/catkin_ws/devel/lib/carla_ad_agent/carla_ad_agent.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/szilard/catkin_ws/src/ros-bridge/carla_ad_agent/src/carla_ad_agent/carla_ad_agent.py", line 16, in <module>
    from basic_agent import BasicAgent  # pylint: disable=relative-import
  File "/home/szilard/catkin_ws/devel/lib/carla_ad_agent/basic_agent.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/szilard/catkin_ws/src/ros-bridge/carla_ad_agent/src/carla_ad_agent/basic_agent.py", line 19, in <module>
    from agent import Agent, AgentState  # pylint: disable=relative-import
  File "/home/szilard/catkin_ws/devel/lib/carla_ad_agent/agent.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/szilard/catkin_ws/src/ros-bridge/carla_ad_agent/src/carla_ad_agent/agent.py", line 16, in <module>
    from misc import is_within_distance_ahead, compute_magnitude_angle  # pylint: disable=relative-import
ImportError: cannot import name 'is_within_distance_ahead' from 'misc' (/home/szilard/catkin_ws/devel/lib/carla_ad_agent/misc.py)
[carla_ad_agent_ego_vehicle-6] process has died [pid 67766, exit code 1, cmd /home/szilard/catkin_ws/devel/lib/carla_ad_agent/carla_ad_agent.py __name:=carla_ad_agent_ego_vehicle __log:=/home/szilard/.ros/log/eb0f636c-3ecd-11eb-a7e1-a755c4dafc1f/carla_ad_agent_ego_vehicle-6.log].
log file: /home/szilard/.ros/log/eb0f636c-3ecd-11eb-a7e1-a755c4dafc1f/carla_ad_agent_ego_vehicle-6*.log
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
Traceback (most recent call last):
  File "/home/szilard/catkin_ws/devel/lib/carla_waypoint_publisher/carla_waypoint_publisher.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/szilard/catkin_ws/src/ros-bridge/carla_waypoint_publisher/src/carla_waypoint_publisher/carla_waypoint_publisher.py", line 35, in <module>
    from agents.navigation.global_route_planner import GlobalRoutePlanner
ModuleNotFoundError: No module named 'agents'
[INFO] [1608036224.831839, 0.000000]: Using ros parameter for spawnpoint: 127.4,195.4,0,0,0,180
[INFO] [1608036224.833276, 0.000000]: listening to server localhost:2000
[INFO] [1608036224.833823, 0.000000]: using vehicle filter: vehicle.tesla.model3
[INFO] [1608036224.834442, 0.000000]: Waiting for CARLA world (topic: /carla/world_info)...
[INFO] [1608036224.855211, 0.000000]: Trying to connect to localhost:2000
[carla_waypoint_publisher-7] process has died [pid 67772, exit code 1, cmd /home/szilard/catkin_ws/devel/lib/carla_waypoint_publisher/carla_waypoint_publisher.py __name:=carla_waypoint_publisher __log:=/home/szilard/.ros/log/eb0f636c-3ecd-11eb-a7e1-a755c4dafc1f/carla_waypoint_publisher-7.log].
log file: /home/szilard/.ros/log/eb0f636c-3ecd-11eb-a7e1-a755c4dafc1f/carla_waypoint_publisher-7*.log

```

[2] [roslaunch-pop-os-106145.log](https://github.com/carla-simulator/ros-bridge/files/5696215/roslaunch-pop-os-106145.log)

joel-mb, 2020-12-23:
Hi @bessszilard ! For what I see in your log, the `agents` module is not being found. Try adding the following line to your `bashrc`:
```sh
export PYTHONPATH=$PYTHONPATH:"<YOUR-CARLA-ROOT-PATH>/PythonAPI/carla"
```

KevinLADLee, 2021-01-02:
Hi @bessszilard,  I had the same problem on Ubuntu 20.04 with ROS noetic. I tried to remove the install command in `carla_ad_agent/CMakeLists.txt`, and it works fine. Maybe this is a problem caused by `catkin_install_python`.

joel-mb, 2021-03-16:
Closing due to no further replies. Please feel free to reopen the issue if the problem persists.

closed 2021-03-16 by joel-mb

Enchantee, 2024-01-22:
Hi I'm stuck at the same situation. Do you know how to solve?
Especially, My Error is ..

File "/mnt/sda/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_waypoint_publisher/src/carla_waypoint_publisher/carla_waypoint_publisher.py", line 25, in <module>
    from agents.navigation.global_route_planner import GlobalRoutePlanner
ModuleNotFoundError: No module named 'agents.navigation'

Please let me know