carla-simulator/ros-bridge #448: TF_REPEATED_DATA warnings

FHof, 2020-12-24:
### Problem description

When I execute `roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch` in a Docker Container with the Rosbridge and ROS noetic installed, many `TF_REPEATED_DATA` warnings appear.
Here are the output messages: [example_ego_out.txt](https://github.com/carla-simulator/ros-bridge/files/5739199/example_ego_out.txt)
Can I safely ignore these warnings?

### Additional information

Rosbridge version: 959039ad9d84d167fe9ef79e69da4a6cfaa3adef
CARLA version: 0.9.10

anavc97, 2020-12-29:
Had this exact same [issue](https://github.com/carla-simulator/carla/issues/3746), which I suspect resulted in some errors running  pygame windows (such as manual_control.py).
I switched from CARLA version 0.9.10 to CARLA 0.9.9 and the warning (and errors) were fixed

FHof, 2020-12-31:
The image in the pygame window looks correct for me.

anavc97, 2021-01-04:
> The image in the pygame window looks correct for me.

The warnings are back but the issues I had were fixed when I switched to CARLA 0.9.9

EsquilliDev, 2021-02-08:
FYI: The issue has to do with [this](https://github.com/ros/geometry2/issues/467).

joel-mb, 2021-03-29:
@FHof @Esquilli @anavc97 Could you please test if the issue is still happening with the latest `master`. Please, take into account that the `master` branch is targeting CARLA 0.9.11

anavc97, 2021-03-29:
@joel-mb I assume you'd want us to test this with the ros-bridge version of 0.9.10. Unfortunately I can't seem to make it work with 0.9.11 in my desktop.

joel-mb, 2021-04-12:
Hi @anavc97, sorry for the delay. I missed your message. The update has been introduced in `master` so you should test it with that branch. Which problem were you facing with `0.9.11`? Right now, `master` is targeting CARLA `0.9.11` so hopefully your problem has disappear now.

On the other hand, I'm still seeing the warning after the updates. It seems that this is only happening for CARLA event sensors (i.e., DVS Camera sensor, Collision sensor and Lane Invasion sensor). I suspect CARLA is sending multiple measurements in the same tick for that sensors under some situations and therefore the ROS bridge is publishing TF data with repeated timestamps. I don't think this is affecting to the simulation as this was already happening with melodic (these warning were added in noetic as stated by @Esquilli).

I will leave this issue open until a good solution for these warnings is found.

anavc97, 2021-04-13:
> Hi @anavc97, sorry for the delay. I missed your message. The update has been introduced in `master` so you should test it with that branch. Which problem were you facing with `0.9.11`? Right now, [...]

@joel-mb thank you for your reply. I can't remember the exact problem with CARLA version `0.9.11`, I think it had something to do with a package called "ainstein radar", which was not being recognized in Noetic, but I'm not sure with the error was. I might open a new issue for that, but it doesn't look like it is related to the ros bridge.

Regarding the warning, I've upgraded to carla-ros-bridge` 0.9.11` and have been running `carla_ros_bridge_with_example_ego_vehicle.launch` but I've not been seeing the warnings, although I did disable a lot of sensors, the DVS included. (I'm running ROS Noetic with Ubuntu 20 and Python3). Hope this helps.