carla-simulator/ros-bridge #457: [ROS2] Problem building ros_compatibility

Michael-Equi, 2021-01-09:
Hello, I am trying to build the ros-bridge for ROS2 foxy on Ubuntu 20.04 but I get the following error:
```
Starting >>> ros_compatibility
--- stderr: ros_compatibility                   
error: error in 'egg_base' option: '../../../build/ros_compatibility' does not exist or is not a directory
---
Failed   <<< ros_compatibility [0.37s, exited with code 1]
```

This is what I ran to get the error:
```bash
# Install Carla Python API 
sudo python3 -m easy_install ~/Carla/PythonAPI/carla/dist/carla-0.9.11-py3.7-linux-x86_64.egg

#setup folder structure
mkdir -p ~/carla-ros-bridge/catkin_ws/src
cd ~/carla-ros-bridge
git clone https://github.com/carla-simulator/ros-bridge.git
cd ros-bridge
git checkout ros2
git submodule update --init
cd ../catkin_ws/src
ln -s ../../ros-bridge
source /opt/ros/foxy/setup.bash
cd ..

# set RMW to FastRTPS
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp

#install required ros-dependencies
rosdep update
rosdep install --from-paths src --ignore-src -r

#build
colcon build
```

Any guidance or help getting this resolved would be greatly appreciated.

james-yoo, 2021-01-10:
@Michael-Equi 
I have solved by setting PYTHONPATH environment variable in ~/.bashrc.
In your case:
export PYTHONPATH=$PYTHONPATH:~/Carla/PythonAPI/carla/dist/carla-0.9.11-py3.7-linux-x86_64.egg

closed 2021-01-15 by Michael-Equi

MisterioRemo, 2021-01-22:
I have similar problem with ros2 dashing. And it was caused by symbolic links.
[Solution](https://answers.ros.org/question/364060/colcon-fails-to-build-python-package-error-in-egg_base/) in Ros answers.