carla-simulator/ros-bridge #467: error from "from ._tf2 import *"

happyOBO, 2021-01-20:
**Problem Description**

I couldn't execute the lunch file because the following error occurred.

```s
$roslaunch carla_ros_bridge carla_ros_bridge_with_rviz.launch
...
Traceback (most recent call last):
  File "/home/obo/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 31, in <module>
    import carla_common.transforms as trans
  File "/home/obo/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_common/src/carla_common/transforms.py", line 18, in <module>
    import tf
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
```

I did the following to solve this problem.

1. add python path : ```export PYTHONPATH=$PYTHONPATH:/home/obo/carla/PythonAPI/carla/dist/carla-0.9.11-py3.7-linux-x86_64.egg```
2. change the python path of `catkin_make` 
  ```s
  catkin_make --cmake-args \
              -DCMAKE_BUILD_TYPE=Release \
              -DPYTHON_EXECUTABLE=/home/obo/anaconda3/envs/carla/bin/python3.7 \
              -DPYTHON_INCLUDE_DIR=/home/obo/anaconda3/envs/carla/include/python3.7m \
              -DPYTHON_LIBRARY=/home/obo/anaconda3/envs/carla/lib/libpython3.7m.so
  ```

how can I solve this? thanks.

YifuTao, 2021-01-30:
I faced the same issue. Does anyone have the solution?

YifuTao, 2021-02-02:
I faced the same issue. The problem is because tf2 requires Python 2 so Python 3 leads to this error.
My solution is:
1. build python 2 egg file if necessary. That can be done by running `make PythonAPI` as stated in [here](https://carla.readthedocs.io/en/latest/build_system/#pythonapi).
2. add python path: `export PYTHONPATH=$PYTHONPATH:/home/yifu/workspace/carla/PythonAPI/carla/dist/carla-0.9.10-py2.7-linux-x86_64.egg`

Make sure that when you run python 2 you got no error when typing `import carla`. Issue related to this is caused by not adding the carla python egg file in your `$PYTHONPATH`.
My `PYTHONPATH` looks something like `/opt/carla-ros-bridge/melodic/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages:/home/yifu/workspace/carla/PythonAPI/carla/dist/carla-0.9.10-py2.7-linux-x86_64.egg`

happyOBO, 2021-02-08:
Thank you. I solved the problem by changing from ubuntu 16.04 (kinetic) to ubuntu 18.04 (melodic).

closed 2021-02-08 by happyOBO