carla-simulator/ros-bridge #468: CARLA python module version 0.9.10 required. Found: 0.9.11

shandaolu, 2021-01-20:
Hello, I met a question when I install ros-brige. How can I solve the problem. 
Carla Version :0.9.11
Does Carla-Ros-bridge only support Carla 0.9.10? 

process[carla_ros_bridge-2]: started with pid [8027]
[INFO] [1611105008.875696, 0.000000]: Trying to connect to localhost:2000
[FATAL] [1611105008.884435, 0.000000]: CARLA python module version 0.9.10 required. Found: 0.9.11
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 8027, exit code 1, cmd /home/alpha/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/alpha/.ros/log/3b8d93d6-5abc-11eb-827e-b42e99a8dab8/carla_ros_bridge-2.log].
log file: /home/alpha/.ros/log/3b8d93d6-5abc-11eb-827e-b42e99a8dab8/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
[carla_ros_bridge-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

LucasY-Y-Y, 2021-01-21:
I had the same issue and what I did was to change the line CARLA_VERSION = "0.9.10" in bridge.py to CARLA_VERSION = "0.9.11". And I was able to launch ros-bridge. However, I'm not sure if it will break other codes. (It works fine for me so far.)

bbrito, 2021-01-25:
@shandaolu we downloaded the wrong Carla version. if you download the correct one (0.9.10) this will be solved.

bbrito, 2021-01-26:
@shandaolu I think you can close this issue.

closed 2021-02-09 by fpasch

martino92, 2022-04-12:
> @shandaolu we downloaded the wrong Carla version. if you download the correct one (0.9.10) this will be solved.

[FATAL] [1649761157.259664, 0.000000]: CARLA python module version 0.9.12 required. Found: 0.9.13
Do you know which version of ROS-bridge is matching with 0.9.13?

martino92, 2022-04-12:
> I had the same issue and what I did was to change the line CARLA_VERSION = "0.9.10" in bridge.py to CARLA_VERSION = "0.9.11". And I was able to launch ros-bridge. However, I'm not sure if it will break other codes. (It works fine for me so far.)

Where is the bridge.py ?