carla-simulator/ros-bridge #469: CARLA Interface has very dark brightness when using roslaunch

pricejiang, 2021-01-20:
[image]
As you can see, the entire CARLA interface has very dark brightness, and I can barely see the car and buildings. 
I used `roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch` to initiate the simulation.  
I tried on both CARLA 0.9.10 and 0.9.11, and end up in the same situation. 
Any ideas on the cause of this issue? 
Thanks!

pricejiang, 2021-01-20:
Problem fixed by modifying camera exposure parameters located in `/opt/carla-ros-bridge/melodic/share/carla_ego_vehicle/config/sensors.json`

closed 2021-01-20 by pricejiang

paolopellegrino, 2021-01-20:
Hi,

may ask how you set those parameters? I still can't get a normal configuration.

Thanks!

pricejiang, 2021-01-20:
> Hi,
> 
> may ask how you set those parameters? I still can't get a normal configuration.
> 
> Thanks!

Hi, 

I changed:
* `exposure_mode` to histogram;
* `exposure_compsensation` to 0.0;
* `exposure_min_bright` to 7.0;
* `exposure_max_bright`to 9.0;

Thanks

fpasch, 2021-01-20:
With the latest version you should be able to simply remove all the parameters. Then the default values are used.

paolopellegrino, 2021-01-21:
There's something strange or something that I'm missing. With the parameters used by @pricejiang it didn't work, neither could remove all the parameters. But if I use the package compiled from source (setting CARLA_VERSION to 0.9.11 in bridge.py) and not the compiled one it works, even if there are differences between the two bridges: isn't this strange or, in fact, I'm missing something?

Thanks

cvcore, 2021-05-15:
The solution from @pricejiang works for me. I'm using `carla-simulator 0.9.11` together with `carla-ros-bridge 0.9.10-1` under ROS melodic. Both packages are installed with `apt install`.