carla-simulator/ros-bridge #478: Depth cloud output perpendicular to the ego vehicle

ammar-n-abbas, 2021-02-09:
Why is the output of the depth cloud perpendicular to the ego vehicle direction. 

[image]

fpasch, 2021-02-09:
From which sensor do you get the depth cloud output?

ammar-n-abbas, 2021-02-09:
@fpasch Depth camera

fpasch, 2021-02-09:
Depth camera of ros-bridge (e.g. instantiated [here](https://github.com/carla-simulator/ros-bridge/blob/master/carla_spawn_objects/config/objects.json#L98) and implemented [here](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/camera.py#L226)) only provides an image with one channel.

ammar-n-abbas, 2021-02-09:
Yes, but the image is then transformed to a depth cloud through depth_cloud_proc by ros which can then be visualized through rviz/rtabmap

fpasch, 2021-02-09:
Could you please try it with the [ros2 branch](https://github.com/carla-simulator/ros-bridge/tree/ros2). And don't get confused by the name, it's the development branch for a hybrid ROS1/ROS2 version of the bridge. It will become the new master in the future.

ammar-n-abbas, 2021-02-09:
Alright, thank you so much :)

ammar-n-abbas, 2021-02-09:
having this error while catkin_make

```
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
sed: can't read msg/CameraImage.msg: No such file or directory
sed: can't read msg/ContourPointSigma.msg: No such file or directory
sed: can't read msg/DeviceStatus.msg: No such file or directory
sed: can't read msg/ErrorWarning.msg: No such file or directory
sed: can't read msg/HostVehicleState2805.msg: No such file or directory
sed: can't read msg/HostVehicleState2806.msg: No such file or directory
sed: can't read msg/HostVehicleState2807.msg: No such file or directory
sed: can't read msg/IbeoDataHeader.msg: No such file or directory
sed: can't read msg/MountingPositionF.msg: No such file or directory
sed: can't read msg/Object2221.msg: No such file or directory
sed: can't read msg/Object2225.msg: No such file or directory
sed: can't read msg/Object2270.msg: No such file or directory
sed: can't read msg/Object2271.msg: No such file or directory
sed: can't read msg/Object2280.msg: No such file or directory
sed: can't read msg/ObjectData2221.msg: No such file or directory
sed: can't read msg/ObjectData2225.msg: No such file or directory
sed: can't read msg/ObjectData2270.msg: No such file or directory
sed: can't read msg/ObjectData2271.msg: No such file or directory
sed: can't read msg/ObjectData2280.msg: No such file or directory
sed: can't read msg/Point2Df.msg: No such file or directory
sed: can't read msg/Point2Di.msg: No such file or directory
sed: can't read msg/Point2Dui.msg: No such file or directory
sed: can't read msg/ResolutionInfo.msg: No such file or directory
sed: can't read msg/ScanData2202.msg: No such file or directory
sed: can't read msg/ScanData2204.msg: No such file or directory
sed: can't read msg/ScanData2205.msg: No such file or directory
sed: can't read msg/ScanData2208.msg: No such file or directory
sed: can't read msg/ScannerInfo2204.msg: No such file or directory
sed: can't read msg/ScannerInfo2205.msg: No such file or directory
sed: can't read msg/ScanPoint2202.msg: No such file or directory
sed: can't read msg/ScanPoint2204.msg: No such file or directory
sed: can't read msg/ScanPoint2205.msg: No such file or directory
sed: can't read msg/ScanPoint2208.msg: No such file or directory
sed: can't read msg/Sigma2D.msg: No such file or directory
sed: can't read msg/Size2D.msg: No such file or directory
sed: can't read msg/TrackedProperties.msg: No such file or directory
sed: can't read msg/UntrackedProperties.msg: No such file or directory
sed: can't read msg/Velocity2D.msg: No such file or directory
CMake Error at /home/ammar/carla-ros-bridge/catkin_ws/build/ros-bridge/astuff_sensor_msgs/ibeo_msgs/cmake/ibeo_msgs-genmsg.cmake:3 (message):
  Could not find messages which
  '/home/ammar/carla-ros-bridge/catkin_ws/src/ros-bridge/astuff_sensor_msgs/ibeo_msgs/msg/CameraImage.msg'
  depends on.  Did you forget to specify generate_messages(DEPENDENCIES ...)?

  Cannot locate message [Time]: unknown package [builtin_interfaces] on
  search path [{{'ibeo_msgs':
  ['/home/ammar/carla-ros-bridge/catkin_ws/src/ros-bridge/astuff_sensor_msgs/ibeo_msgs/msg'],
  'std_msgs': ['/opt/ros/melodic/share/std_msgs/cmake/../msg']}}]
Call Stack (most recent call first):
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  ros-bridge/astuff_sensor_msgs/ibeo_msgs/CMakeLists.txt:73 (generate_messages)

-- Configuring incomplete, errors occurred!
See also "/home/ammar/carla-ros-bridge/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ammar/carla-ros-bridge/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

ammar-n-abbas, 2021-02-09:
Deleted the astuff_sensor_msgs just for testing. The depth cloud has now been tilted at an angle 

[image]

ammar-n-abbas, 2021-02-12:
Fixed using:

 synchronous_mode: true in ~/carla-ros-bridge/ros-bridge/carla_ros_bridge/config/settings.yaml

and editing rgbd_mapping.launch file to:

```
<!-- -->
<launch>
 
  <!-- Choose visualization -->
  <arg name="rviz"                    default="false" />
  <arg name="rtabmapviz"              default="true" /> 
  
  <!-- Localization-only mode -->
  <arg name="localization"            default="false"/>
  
  <!-- Corresponding config files -->
  <arg name="rtabmapviz_cfg"          default="~/.ros/rtabmap_gui.ini" />
  <arg name="rviz_cfg"                default="$(find rtabmap_ros)/launch/config/rgbd.rviz" />
 
  <arg name="pi/2" value="1.5707963267948966" />
  <node pkg="tf" type="static_transform_publisher" name="base_link" args="0 0 2 -$(arg pi/2) 0 -$(arg pi/2) ego_vehicle ego_vehicle/depth_front 0" />

  <node pkg="tf" type="static_transform_publisher" name="base_link2" args="0 0 2 -$(arg pi/2) 0 -$(arg pi/2) ego_vehicle ego_vehicle/rgb_front 0" />

  <arg name="frame_id"                default="ego_vehicle"/>   <!-- Fixed frame id, you may set "base_link" or "base_footprint" if they are published -->
  <arg name="database_path"           default="~/.ros/rtabmap.db"/>
  <arg name="rtabmap_args"            default="-- delete_db_on_start"/>              <!-- delete_db_on_start, udebug -->
  <arg name="launch_prefix"           default=""/>              <!-- for debugging purpose, it fills launch-prefix tag of the nodes -->
  <arg name="approx_sync"             default="false"/>         <!-- if timestamps of the input topics are not synchronized -->
   
  <arg name="rgb_topic"               default="/carla/ego_vehicle/rgb_front/image" />
  <arg name="depth_registered_topic"  default="/carla/ego_vehicle/depth_front/image" />
  <arg name="camera_info_topic"       default="/carla/ego_vehicle/rgb_front/camera_info" />
  <arg name="compressed"              default="false"/>
   
  <arg name="subscribe_scan"          default="false"/>         <!-- Assuming 2D scan if set, rtabmap will do 3DoF mapping instead of 6DoF -->
  <arg name="scan_topic"              default="/carla/ego_vehicle/depth_front/scan"/>
  
  <arg name="subscribe_scan_cloud"    default="true"/>         <!-- Assuming 3D scan if set -->
  <arg name="scan_cloud_topic"        default="/carla/ego_vehicle/lidar"/>
  
  <arg name="visual_odometry"         default="false"/>          <!-- Generate visual odometry -->
  <arg name="odom_topic"              default="/carla/ego_vehicle/odometry"/>         <!-- Odometry topic used if visual_odometry is false -->
  <arg name="odom_frame_id"           default=""/>              <!-- If set, TF is used to get odometry instead of the topic -->
   

  <arg name="namespace"               default="rtabmap"/>
  <arg name="wait_for_transform"      default="0.2"/>
  
  <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
    <arg name="rtabmapviz"              value="$(arg rtabmapviz)" /> 
    <arg name="rviz"                    value="$(arg rviz)" />
    <arg name="localization"            value="$(arg localization)"/>
    <arg name="gui_cfg"                 value="$(arg rtabmapviz_cfg)" />
    <arg name="rviz_cfg"                value="$(arg rviz_cfg)" />
  
    <arg name="frame_id"                value="$(arg frame_id)"/>
    <arg name="namespace"               value="$(arg namespace)"/>
    <arg name="database_path"           value="$(arg database_path)"/>
    <arg name="wait_for_transform"      value="$(arg wait_for_transform)"/>
    <arg name="rtabmap_args"            value="$(arg rtabmap_args)"/>  
    <arg name="launch_prefix"           value="$(arg launch_prefix)"/>          
    <arg name="approx_sync"             value="$(arg approx_sync)"/>

    <arg name="rgb_topic"               value="$(arg rgb_topic)" />
    <arg name="depth_topic"             value="$(arg depth_registered_topic)" />
    <arg name="camera_info_topic"       value="$(arg camera_info_topic)" />
    <arg name="compressed"              value="$(arg compressed)"/>                                                                                
   
    <arg name="subscribe_scan"          value="$(arg subscribe_scan)"/>
    <arg name="scan_topic"              value="$(arg scan_topic)"/>
  
    <arg name="subscribe_scan_cloud"    value="$(arg subscribe_scan_cloud)"/>
    <arg name="scan_cloud_topic"        value="$(arg scan_cloud_topic)"/>
   
    <arg name="visual_odometry"         value="$(arg visual_odometry)"/>          
    <arg name="odom_topic"              value="$(arg odom_topic)"/>    
    <arg name="odom_frame_id"           value="$(arg odom_frame_id)"/>         
    <arg name="odom_args"               value="$(arg rtabmap_args)"/>
  </include>

</launch>

```

closed 2021-03-16 by joel-mb

ajay1606, 2021-12-06:
> having this error while catkin_make
>
> ```
> -- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
> sed: can't read msg/CameraImage.msg: No such file or directory
> sed: can't read [...]

@ammar-n-abbas Hello Ammar, I also stuck with same Cmake error one that you mentioned it here, May I know how you would able to fix this !

Regards,
Ajay

ammar-n-abbas, 2021-12-06:
Hi @ajay1606 

Have you tried the fixes that were mentioned below this error message?

1 Deleted the astuff_sensor_msgs
2 synchronous_mode: true in ~/carla-ros-bridge/ros-bridge/carla_ros_bridge/config/settings.yaml and editing rgbd_mapping.launch file

ajay1606, 2021-12-06:
@ammar-n-abbas 
Thanks for your quick response. In my case, I am not using CARLA, instead, I am working with IBEO lidars. Actually CMAKE error message one that you mentioned is because of Ibeo lidar ROS driver.  To work with IBEO ROS driver those mgs files are necessary. 

Actually, there were no official ROS support provided to IBEO lidars, i posted here after looking at same error message with CARLA sim. I thought CARLA sim it would have fixed !

ammar-n-abbas, 2021-12-06:
@ajay1606 oh okay, then I do not know about the fix, sorry...