carla-simulator/ros-bridge #479: Carla ros bridge doesn't find python

Johanfanas, 2021-02-09:
Hi,

I downloaded the carla simulator and created a ROS workspace for the ros-bridge. I have everything setup properly but I am getting this error when launching the ros bridge "/usr/bin/env: ‘python’: No such file or directory". I am using Ubuntu 20.04. In this OS, python is not located in /usr/bin/env, so how do I tell the ros bridge where to find python ?

[image]

fpasch, 2021-02-10:
Please try:
```
sudo apt install python-is-python3
```

Johanfanas, 2021-02-10:
Thanks a lot, that helps. Now I am getting an error saying the ros bridge I have requieres Carla 0.9.10, but I saw the Carla repo and it says the Carla version 0.9.10 is only compatible for Ubuntu 18.04. Any suggestions ?

hofbi, 2021-02-10:
0.9.10 also works on 20.04, but you need to download the release directly from github: https://github.com/carla-simulator/carla/releases/tag/0.9.10.1

The issue https://github.com/carla-simulator/carla/issues/3441 tracks the progress for supporting the releases on the carla ppa.

joel-mb, 2021-03-16:
Closing the issue as the latest CARLA version has been added to the apt repo for focal.
Please, feel free to reopen the issue if the problem persists.

closed 2021-03-16 by joel-mb