carla-simulator/ros-bridge #483: roslaunch pcl_recorder pcl_recorder.launch with manual drive?

athulg, 2021-02-22:
Is it possible to record pcd by using manual_control instead of Autopilot? 

Thanks!

athulg, 2021-02-22:
I was wondering how to save the data in" #The captured point clouds are saved to /tmp/pcl_capture directory."? I'm using manual control instead of autopilot and I'm not sure how to end a session once I've covered driving around the whole map and save data in tmp/pcl_capture ?

fpasch, 2021-02-24:
I don't get your question.

If you use [carla_manual_control](https://github.com/carla-simulator/ros-bridge/tree/master/carla_manual_control) you can either drive around manually after pressing <B> or press <P> to enable autopilot.

While driving the [pcl_recorder](https://github.com/carla-simulator/ros-bridge/tree/master/pcl_recorder) needs to be running. Then the captures get stored in `/tmp/pcl_capture`.

athulg, 2021-02-25:
@fpasch I’m sorry for not being clear, for some reason the recording didn’t start with pcl_recorder and hence the data was not saved. But, when I tried it in a different system it worked perfectly.

closed 2021-02-25 by athulg