carla-simulator/ros-bridge #494: No `carla-ros-bridge` in Ubuntu 20.04

Alby407, 2021-03-04:
It seems there is no package named `carla-ros-bridge` in Ubuntu 20.04 at the moment. After following the steps, apt does not find the package.

hofbi, 2021-03-04:
Duplicate of #415

Alby407, 2021-03-04:
It's not a duplicate, since the simulator itself is in the repo. But the ROS bridge itself is not.

Alby407, 2021-03-04:
Okay, I will follow the dev setup then. 
Are the changes to the ROS bridge for version 0.9.11 stated on the Carla website in the master branch?

Alby407, 2021-03-07:
Ros-bridge still needs to be updated to CARLA version 0.9.11.

joel-mb, 2021-03-16:
Hi @Alby407! You are right, there is no debian package for the `ros-bridge` in focal yet. I would recommend using the `dev` instructions. Regarding the CARLA version, the latest ros-bridge release is for 0.9.10.1 as stated in the README. However, we've been testing 0.9.11 and seems that everything works well. To use 0.9.11 just modify this line in `bridge.py` (https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/bridge.py#L54) with the proper version