carla-simulator/ros-bridge #502: ('Failed to localize! : When I try to roslaunch carla_autoware_bridge carla_autoware_bridge.launch

AthulGangadharan, 2021-03-08:
Hello, I'm trying to launch the carla_autoware_bridge with 
`  
  <include file="$(find carla_waypoint_publisher)/launch/carla_waypoint_publisher.launch" >`

for a custom map created in RoadRunner 

`
[INFO] [1615200695.516626, 200.698360]: CARLA world available. Spawn ego vehicle...
WARNING: Traffic sign 656 overlaps a driving lane. Moving out of the road... 
WARNING: Failed to find suitable place for signal. 
WARNING: Traffic sign 655 overlaps a driving lane. Moving out of the road... 
WARNING: Failed to find suitable place for signal. 
WARNING: Traffic sign 659 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 657 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 654 overlaps a driving lane. Moving out of the road... 
[INFO] [1615200695.674896, 202.198360]: Ego vehicle changed.
[INFO] [1615200695.676381, 202.198360]: Calculating route to x=97.3952484131, y=-28.0436782837, z=4.0
[INFO] [1615200695.706405, 202.398360]: Created EgoVehicle(id=341, parent_id=None, type=vehicle.tesla.model3, prefix=ego_vehicle, attributes={'number_of_wheels': '4', 'role_name': 'ego_vehicle', 'object_type': '', 'sticky_control': 'true'})
[INFO] [1615200695.707654, 202.398360]: Created ObjectSensor(parent_id=341, prefix=ego_vehicle/objects)
WARNING: Traffic sign 656 overlaps a driving lane. Moving out of the road... 
WARNING: Failed to find suitable place for signal. 
WARNING: Traffic sign 655 overlaps a driving lane. Moving out of the road... 
WARNING: Failed to find suitable place for signal. 
WARNING: Traffic sign 659 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 657 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 654 overlaps a driving lane. Moving out of the road... 
('Failed to localize! : ', 'Road id : ', 34, 'Section id : ', 0, 'Lane id : ', -4, 'Location : ', 30.977561950683594, 182.38272094726562)
('Failed to localize! : ', 'Road id : ', 34, 'Section id : ', 0, 'Lane id : ', -4, 'Location : ', 31.56435775756836, 181.54159545898438)
('Failed to localize! : ', 'Road id : ', 34, 'Section id : ', 1, 'Lane id : ', -4, 'Location : ', 32.97539520263672, 179.49154663085938)
('Failed to localize! : ', 'Road id : ', 34, 'Section id : ', 1, 'Lane id : ', -4, 'Location : ', 33.551475524902344, 178.64309692382812)
.....`

Any suggestion on what I might be doing wrong?

Thanks

Axel1092, 2021-03-16:
Hi @AthulGangadharan, about the `WARNING: Traffic sign 656 overlaps a driving lane. Moving out of the road... WARNING: Failed to find suitable place for signal.` messages they are not errors. They just mean that some traffic lights or signs are placed over the road and the simulator tries to displace them to a close but visually better location. Still, this should only have visual effects (e.g. using CARLA RGB cameras) but not on the vehicle behavior (traffic sign detection should work the same).

joel-mb, 2021-03-16:
Hi @AthulGangadharan, the `('Failed to localize! : ', 'Road id : ', 34, 'Section id : ', 0, 'Lane id : ', -4, 'Location : ', 30.977561950683594, 182.38272094726562)` messages are related to the CARLA global route planner.
Which CARLA version are you using? Can you send us the OpenDrive file that you are using?

AthulGangadharan, 2021-03-16:
Hi @Axel1092, thank you for the reply, these warnings are then leading to publish no waypoints. And then I can't set a goal in Autoware. I'm also trying to do this without using any of the vector maps. I can send you the detailed error log later as well.

AthulGangadharan, 2021-03-16:
Hey @joel-mb, I'm using CARLA 0.9.10, Is there a way I can privately send you the OpenDrive file?

joel-mb, 2021-03-16:
@AthulGangadharan Ok! Please, feel free to reopen the issue or send us your OpenDrive file if the problem with the CARLA global planner persists. To privately contact me you can use discord.

closed 2021-03-16 by joel-mb