carla-simulator/ros-bridge #505: LiDAR point cloud inaccurate/missing for trees/buildings in Town 02

YifuTao, 2021-03-12:
Hi, thanks for the useful ros-bridge package!

I was collecting data in Town02 by driving a vehicle with lidar attached. I was projecting lidar onto the image plane, and I found the lidar point is not accurate. 

**Example 1**
In Rviz, we see only some of the LiDAR points or even no points on the building seen in the image. In this example there are buildings on the right in the rgb image, but not points of that in the point cloud. (while trees further away is scanned which are on the top right of the left point cloud image)

[image]
However, the ground truth depth camera gives perfect result. The house missing in LiDAR is shown in the depth image as shown above. 

The missing lidar points cause the fact that in the image overlay we only see partial lidar points / no lidar points. (Apparently the projection is correct as for most parts in the image. The lidar points on the poles are accurate)
[image]
[image]
[image]
Also, the tree lidar points is not accurate. I understand this is difficult to model, but in the depth camera the tree is segmented very well, while the lidar does not overlay well.
[image]

It seems the LiDAR rendering and the 3D model of the buildings is causing this problem. 

Can you help on this? Thanks a lot for the effort!

joel-mb, 2021-03-16:
Hi @YifuTao! Thank you for the detailed description.
Which version of CARLA are you using?

YifuTao, 2021-03-16:
Hi, I am using CARLA 0.9.10 quick start version downloaded from https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.10.tar.gz  on Ubuntu 18.04 .

joel-mb, 2021-03-16:
Hi @YifuTao!
I've been talking with the core and art team about your issues.

* **Missing points in buildings:**  This is a known issue. Some towns have missing colliders for some buildings. The art team is working to fix this in the following CARLA releases. I would recommend you using `Town10HD` for your testing instead of the older towns. This issue should disappear at `Town10HD`. Also, you could also test using the latest CARLA release (https://github.com/carla-simulator/carla/releases/tag/0.9.11). The ROS bridge is currently targeting 0.9.10 but we haven't seen any issue while testing it with 0.9.11. To use 0.9.11 just change this line in `bridge.py` to set the proper version (https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/bridge.py#L54)
* **Inaccurate point cloud in trees**: This is also a known limitation. The `raycast` used to build the point cloud is using the original model without taking into account the animations of the trees. I'm afraid we cannot do anything to improve this situation right now.

Hope this helps.

YifuTao, 2021-03-16:
Thank you very much for the detailed answer @joel-mb !
The inaccurate tree points in LiDAR is a bigger problem for us compared to the missing points...
May I ask why is the depth image perfect and takes the animation into account, but LiDAR does not? In theory, if the LiDAR has no noise, its re-projection onto the image should correspond to the pixels in depth image.

DSantosO, 2021-03-31:
Hello, 
Sorry for the late response.
The difference between the camera and the lidar is that the Lidar is performed using UE4 raycasts (internally uses PhysX to do that) and does not take into account the animation of the trees. 
For now, there is no much that we can do about that, sorry.

closed 2022-07-28 by YifuTao