carla-simulator/ros-bridge #511: Camera TF orientation problem

KevinLADLee, 2021-03-24:
Hi! I tested the 0.9.11 version of carla with the ros-bridge master branch and found a problem. the camera visualization option in Rviz does not display the point cloud to RGB image projection properly. This problem is not present in the 0.9.10.1 version of ros-bridge. It seems to be caused by the wrong orientation of the camera coordinate system. Hopefully, this issue can be attended to.
[image]

KevinLADLee, 2021-03-24:
FYI
[image]

joel-mb, 2021-03-29:
Hi @KevinLADLee. Thanks for reporting the issue. I've introduced the fixes in the following branch: https://github.com/carla-simulator/ros-bridge/tree/joel-mb/fix_camera_tf. Could you confirm that this is fixing the issue?

KevinLADLee, 2021-03-29:
[image]
I think it works well. Thanks a lot!

PR #516 (2021-04-06): Fix camera TF orientation
Fixes #511

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/ros-bridge/516)
closed 2021-04-06 by joel-mb