carla-simulator/ros-bridge #515: exit code -11

cslxiao, 2021-03-27:
I have tried Carla 0.9.10 and 0.9.11 with python 3.6 / 3.7 but constantly got the error with exit code -11.
here are the output:
```
... logging to /home/omnisky/.ros/log/2773afaa-8edf-11eb-bf81-0c9d92c5f85e/roslaunch-omnisky-30443.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://omnisky:41195/

SUMMARY
========

PARAMETERS
 * /carla_manual_control_ego_vehicle/role_name: ego_vehicle
 * /carla_ros_bridge/ego_vehicle_role_name: ["hero", "ego_veh...
 * /carla_ros_bridge/fixed_delta_seconds: 0.05
 * /carla_ros_bridge/host: localhost
 * /carla_ros_bridge/passive: 
 * /carla_ros_bridge/port: 2000
 * /carla_ros_bridge/synchronous_mode: True
 * /carla_ros_bridge/synchronous_mode_wait_for_vehicle_control_command: False
 * /carla_ros_bridge/timeout: 10
 * /carla_ros_bridge/town: 
 * /carla_spawn_objects_omnisky_30443_7980580604645810429/objects_definition_file: /home/omnisky/cat...
 * /carla_spawn_objects_omnisky_30443_7980580604645810429/spawn_point_ego_vehicle: 
 * /carla_spawn_objects_omnisky_30443_7980580604645810429/spawn_sensors_only: False
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /set_initial_pose_ego_vehicle/control_id: control
 * /set_initial_pose_ego_vehicle/role_name: ego_vehicle
 * /use_sim_time: True

NODES
  /
    carla_manual_control_ego_vehicle (carla_manual_control/carla_manual_control.py)
    carla_ros_bridge (carla_ros_bridge/bridge.py)
    carla_spawn_objects_omnisky_30443_7980580604645810429 (carla_spawn_objects/carla_spawn_objects.py)
    set_initial_pose_ego_vehicle (carla_spawn_objects/set_initial_pose.py)

auto-starting new master
process[master]: started with pid [30453]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2773afaa-8edf-11eb-bf81-0c9d92c5f85e
process[rosout-1]: started with pid [30464]
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [30471]
process[carla_spawn_objects_omnisky_30443_7980580604645810429-3]: started with pid [30472]
process[set_initial_pose_ego_vehicle-4]: started with pid [30473]
process[carla_manual_control_ego_vehicle-5]: started with pid [30474]
pygame 2.0.1 (SDL 2.0.14, Python 3.7.4)
Hello from the pygame community. https://www.pygame.org/contribute.html
[INFO] [1616837467.741209, 0.000000]: Trying to connect to localhost:2000
[INFO] [1616837467.832994, 0.000000]: Loading town '' (previous: 'Town01').
[INFO] [1616837469.898767, 0.000000]: synchronous_mode: True
[INFO] [1616837469.901387, 0.000000]: fixed_delta_seconds: 0.05
[INFO] [1616837469.950745, 0.000000]: Parameters:
[INFO] [1616837469.953495, 0.000000]:   host: localhost
[INFO] [1616837469.955568, 0.000000]:   port: 2000
[INFO] [1616837469.957454, 0.000000]:   timeout: 10
[INFO] [1616837469.959655, 0.000000]:   passive: 
[INFO] [1616837469.961838, 0.000000]:   synchronous_mode: True
[INFO] [1616837469.963768, 0.000000]:   synchronous_mode_wait_for_vehicle_control_command: False
[INFO] [1616837469.965951, 0.000000]:   fixed_delta_seconds: 0.05
[INFO] [1616837469.967958, 0.000000]:   town: 
[INFO] [1616837469.969903, 0.000000]:   ego_vehicle: {'role_name': '["hero", "ego_vehicle", "hero0", "hero1", "hero2", "hero3", "hero4", "hero5", "hero6", "hero7", "hero8", "hero9"]'}
[INFO] [1616837470.091790, 0.000000]: Created Spectator(id=458)
[INFO] [1616837470.093943, 0.000000]: Created TrafficLight(id=459)
[INFO] [1616837470.096778, 0.000000]: Created TrafficLight(id=460)
[... 51 lines cut ...]
[INFO] [1616837471.471358, 2.024743]: Created EgoVehicle(id=631)
[INFO] [1616837471.475062, 2.024743]: Object (type='sensor.other.radar', id='radar_front') spawned successfully as 636.
[INFO] [1616837471.482520, 2.024743]: Created RgbCamera(id=632)
[INFO] [1616837471.482981, 2.024743]: Object (type='sensor.camera.semantic_segmentation', id='semantic_segmentation_front') spawned successfully as 637.
[INFO] [1616837471.490701, 2.024743]: Created RgbCamera(id=633)
[INFO] [1616837471.493894, 2.024743]: Object (type='sensor.camera.depth', id='depth_front') spawned successfully as 638.
[INFO] [1616837471.739389, 2.024743]: Object (type='sensor.camera.dvs', id='dvs_front') spawned successfully as 639.
[WARN] [1616837471.841403, 2.074743]: Spawning process has been interrupted. There might be actors that have not been destroyed properly
[INFO] [1616837471.844231, 2.074743]: Destroying spawned objects...
[WARN] [1616837471.848826, 2.074743]: Could not call destroy service on objects, the ros bridge is probably already shutdown
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 30471, exit code -11, cmd /home/omnisky/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/omnisky/.ros/log/2773afaa-8edf-11eb-bf81-0c9d92c5f85e/carla_ros_bridge-2.log].
log file: /home/omnisky/.ros/log/2773afaa-8edf-11eb-bf81-0c9d92c5f85e/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
[carla_manual_control_ego_vehicle-5] killing on exit
[set_initial_pose_ego_vehicle-4] killing on exit
[carla_spawn_objects_omnisky_30443_7980580604645810429-3] killing on exit
[carla_ros_bridge-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

joel-mb, 2021-03-29:
Hi @cslxiao!
From your console output, I see that you are using ROS melodic. ROS melodic is targeting Python2. I would recommend using ROS noetic (Ubuntu focal) to execute the bridge with Python3.

cslxiao, 2021-03-29:
Thanks @joel-mb .
But I successfully run them on another pc with all the same setting (i.e., ubuntu18.04 and ros melodic with python3)

joel-mb, 2021-04-12:
@cslxiao sorry for the delay in my response. Did you manage to fix the issue?
I would say that this error is related to your ROS melodic-Python3 configuration and not to the ROS bridge itself. Were you using the ROS bridge in that other environment in which you manage to make it work?

stepan-antonyuk, 2023-07-10:
Hello, I have the same issue. It worked but now it gives exit code -11 every time I try to run it. I am using ros2

Here is what I get:
`santonyuk@justingpu:~/Projects/navigator$ ros2 launch carla.launch.py 
[INFO] [launch]: All log files can be found below /home/santonyuk/.ros/log/2023-07-10-15-18-26-804056-justingpu-769698
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [carla_spawn_objects-2]: process started with pid [769701]
[INFO] [bridge-1]: process started with pid [769699]
[INFO] [liaison_node-3]: process started with pid [769703]
[INFO] [rosbridge_websocket-4]: process started with pid [769705]
[INFO] [robot_state_publisher-5]: process started with pid [769707]
[INFO] [lidar_processing_node-6]: process started with pid [769709]
[INFO] [gnss_processing_node-7]: process started with pid [769711]
[robot_state_publisher-5] [WARN] [1689020307.472845637] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-5] [INFO] [1689020307.476401347] [robot_state_publisher]: got segment base_link
[robot_state_publisher-5] [INFO] [1689020307.476417747] [robot_state_publisher]: got segment hero
[robot_state_publisher-5] [INFO] [1689020307.476422857] [robot_state_publisher]: got segment model_link
[rosbridge_websocket-4] [INFO] [1689020307.949480970] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9091
[ERROR] [bridge-1]: process has died [pid 769699, exit code -11, cmd '/home/share/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge --ros-args -r __node:=carla_ros_bridge --params-file /tmp/launch_params_i25nf6wy --params-file /tmp/launch_params_1bwhdl5f --params-file /tmp/launch_params_0jc72yml --params-file /tmp/launch_params_cjpgfi_n --params-file /tmp/launch_params_okcgd9hc --params-file /tmp/launch_params_xwycbwxz --params-file /tmp/launch_params_eskyn9a2 --params-file /tmp/launch_params_zqz0h_h3 -r /carla/hero/lidar:=/lidar/fused -r /carla/hero/rgb_center/image:=/cameras/camera0'].
[ERROR] [liaison_node-3]: process has died [pid 769703, exit code -11, cmd '/home/santonyuk/Projects/navigator/install/carla_interface/lib/carla_interface/liaison_node --ros-args'].`