carla-simulator/ros-bridge #521: The problem of calculating the wheel position in VehicleInfo

KevinLADLee, 2021-04-09:
Hi, My recent research work uses the position of the wheels in VehicleInfo to calculate some parameters of the ego vehicle. But I found a problem with the way the wheel positions are calculated. As shown in the code below. 
https://github.com/carla-simulator/ros-bridge/blob/8f34e1c60a1006b0719dadf5af1f94f8222a5f98/carla_ros_bridge/src/carla_ros_bridge/ego_vehicle.py#L145-L150

This calculation can only work if there is no rotation at the spawn point. So I tried to solve this problem by using the coordinate transformation function of carla python api.

KevinLADLee, 2021-04-09:
#522

joel-mb, 2021-04-12:
Hi @KevinLADLee! Thank you for reporting the issue. Your PR has been merged.

closed 2021-04-12 by joel-mb